Part-DB/Part-DB-server #549: API key

jbtronics, 2024-03-08:
### Discussed in https://github.com/Part-DB/Part-DB-server/discussions/548

<div type='discussions-op-text'>

<sup>Originally posted by **rdalen** March  8, 2024</sup>
I generated a API key for KiCad with an expiration date of 01-01-2100 - but now i get an "Internal Server Error (Status 500)" when i try to open the User settings menu again. Did I perhaps enter an illegal date for Part-DB? Is there a backdoor to fix this?
The Docker error log says;
[error log.md](https://github.com/Part-DB/Part-DB-server/files/14531302/error.log.md)
</div>

joelvonrotz, 2024-03-08:
Have you given all the API rights to the admin?
And did you correctly fill out the `kicad_httplib`?

[image]

jbtronics, 2024-03-09:
Year 2038 bug on 32-bit now get at least somehow prevented, by forbidding users to choose dates after 2038.

closed 2024-03-09 by jbtronics