Part-DB/Part-DB-server #554: Mass editing of parts

Tropaion, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
Hello,
a feature for mass editing parts would be really nice.
It's especially usefull for new functions like EDA information.
For example editing the eda information of all of my 0603 MLCCs.

**Describe the solution you'd like**
When selecting multiple parts in an list, a button "Edit Selected" where values of all parts can be overwritten.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

isaacolsen94, 2024-03-22:
+1 to this!

d-buchmann, 2024-03-26:
There is already a feature to change certain fields (category, footprint, manufacturer, unit of measure) and boolean values (review needed, favorite) of all selected items.
This may be extenable to EDA information, as well as maybe adding tags or other rather simple data structures.

dtk1985, 2025-01-22:
+1 to bulk editing improvements. for example: i would love to set the stock qty of a lot of parts at the same time (capacitor assortment arrived and it contains 36 capacitors, 20 pcs each)

m-klecka, 2025-03-04:
I would also love this. Mass editing available stock and changing location of parts would be great.

spaceac23, 2025-05-03:
+1
I wanted to move several parts into a new category. It would be very helpful to have an option for that.