Part-DB/Part-DB-server #559: Error 500 when cloning a part with a preview image without an associated attachment

RaptorDE, 2024-03-09:
I have found a bug. If you clone a part that has an image in the attachments as a preview image and you delete the image before cloning, the error 500 is displayed.

The reason for this is that the selection field for the preview image is not cleared when the attachment is deleted. When saving the part
, the preview image referenced in the selection field is then missing. If you empty this field manually, no error occurs.

The error occurs with Docker version 0.10.7 and the online demo. For example with this part:
https://demo.part-db.de/en/part/1535/info#attachments

[image]

RaptorDE, 2024-03-09:
context={"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"} 
level_name=DEBUG channel=security 
datetime=2024-02-24T21:19:16.999271+00:00 
extra={"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"****"}, 
"url":"/de/part/861/clone","ip":"172.18.0.7","http_method":"POST","server":"****","referrer":"https://****/de/part/861/clone"}{"message":"Uncaught 
PHP Exception Doctrine\\ORM\\ORMInvalidArgumentException: \"A new entity 
was found through the relationship 
'App\\Entity\\Parts\\Part#master_picture_attachment' that was not 
configured to cascade persist operations for entity: rc_part_image_861. 
To solve this issue: Either explicitly call EntityManager#persist() on 
this unknown entity or configure cascade persist this association in the 
mapping for example @ManyToOne(..,cascade={\"persist\"}).\" at 
ORMInvalidArgumentException.php line 
103","context":{"exception":{"class":"Doctrine\\ORM\\ORMInvalidArgumentException","message":"A 
new entity was found through the relationship 
'App\\Entity\\Parts\\Part#master_picture_attachment' that was not 
configured to cascade persist operations for entity: rc_part_image_861. 
To solve this issue: Either explicitly call EntityManager#persist() on 
this unknown entity or configure cascade persist this association in the 
mapping for example 
@ManyToOne(..,cascade={\"persist\"}).","code":0,"file":"/var/www/html
/vendor/doctrine/orm/src/ORMInvalidArgumentException.php:103","trace":["/var/www/html/vendor/doctrine/orm/src/UnitOfWork.php:3845","/var/www/html/vendor/doctrine/orm/src/UnitOfWork.php:417","/var/www/html/vendor/doctrine/orm/src/EntityManager.php:403","/var/www/html/src/Controller/PartController.php:328","/var/www/html/src/Controller/PartController.php:221","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:181","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"]}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2024-02-24T21:19:17.071665+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"****"},"url":"/de/part/861/clone","ip":"172.18.0.7","http_method":"POST","server":"****","referrer":"https://****/de/part/861/clone"}}

jbtronics, 2024-03-10:
Should be fixed with latest commit

closed 2024-03-10 by jbtronics

RaptorDE, 2024-03-10:
Thank you very much for the really quick fix. 

I found a other problem with the preview image when cloning. If you clone a part with a preview image and delete the existing preview image from the attachment and add a new image as an attachment, it is not automatically used as a preview image. If you create a new part, the first image in the attachment is added as a preview image.