Part-DB/Part-DB-server #560: Storage sub postions creation does not work

henrikkoch, 2024-03-10:
**Describe the bug**
The helps says that you can create sub positions for storage locations using "->"  and it shows "+" but it does not create any sub locations but only new top locations

If I f,ex has a storage box "PartDB Box1" and want to add a sub position in that box like "A2" then writing the location for the box I already has create PartDB Box1 -> A2 does not create a new sub position but a new top level position called A2

Using version 1.11.2 version

jbtronics, 2024-03-16:
I can not reproduce this issue. If I copy paste "PartDB Box1 -> A2" into to the storage location field of stocks on the part edit page, it creates a new sub location like desired on save

henrikkoch, 2024-03-18:
If I write the whole new string ex.   "Mechanincs -> Connectors -> SMD"
then iit works
But if "Mechanic-Connectors" were already there and I TAB'ed me through
this to show it and just added "-> SMD" in the end of this then it doesn't
work in my end

/Henrik

Den lør. 16. mar. 2024 kl. 18.50 skrev Jan Böhmer ***@***.***

>:

> I can not reproduce this issue. If I copy paste "PartDB Box1 -> A2" into
> to the storage location field of stocks on the part edit page, it creates a
> new sub location like desired on save

jbtronics, 2024-04-28:
That is the way the select picker library works. After you selected an element, the old input gets cleared. That is also shown in the "Plus Element XY" row shown on top, how the new element will be named, so I dont think it is such a big problem.

This is not really easy fixable.

jbtronics, 2024-08-24:
This is now possible (or at least highly improved), see issue #666

closed 2024-08-24 by jbtronics