Part-DB/Part-DB-server #561: Bulk storage create feature

henrikkoch, 2024-03-10:
When creating new locations for storage I add a next box. I'm using the type AIDEtelk boxes. This has a top and a button part of that box

the top part has 12 columns and 6 rows. 

So the feature I'm asking for is a thing to create sub locations to that box in one creation like

AideTek box2, Top part:   Rows 6 [numbers] , Columns 12 [Letters]  creates: A1, A2, A3, A4, A5, A6, B1.............L4 of sub positions to that box

jbtronics, 2024-03-11:
You should be able to somehow use the mass creation feature for that. You still need to type it out, but that should be easier than create all locations manually and you can use some copy paste (and reuse it for later locations).

```
Box2
  A
    A1
    A2
    A3
 B
   B1
   B2
   B3
 C
   C1
   C2
   C3
````

henrikkoch, 2024-03-11:
Thanks that's pretty smart and helps me instead of all that manual work
:-)  Thanks

/Henrik

Den man. 11. mar. 2024 kl. 11.45 skrev Jan Böhmer ***@***.***

>:

> You should be able to somehow use the mass creation feature for that. You
> still need to type it out, but that should be easier than create all
> locations manually and you can use some copy paste (and [...]

closed 2025-09-07 by jbtronics