Part-DB/Part-DB-server #563: Decouple time/date format from language selection

Janub, 2024-03-11:
I think it would be very useful, if you could select your preferred date and time format independently of the UI language. With the current system, selecting English as the UI language automatically also results in a 12h time format and the american MM/DD/YYYY date format.
Ideally you could chose a date and time format on your user settings page, where Theme, Timezone, etc. are configured as well.

Janub, 2024-03-15:
I tried your suggestion and it mostly solves the issue for me. There are, however, two minor problems:
1. The date is shown as DD/MM/YYYY with en_DE, but with just de it is shown as DD.MM.YYYY. It is not a big issue, but still interesting to see as apparently not all of the date formatting is done in the same way.
2. Some string translations are missing when using the en_DE locale. So far I only noticed the search placeholder "search.placeholder" and the indicator for adding a new catecory/footprint/etc. "entity.select.group.not_added_to_DB"
Overall I would still love to have the option to choose date format, time format, language (and perhaps decimal separator) as a user setting, but the manual choice of locale is a good workaround for now.

jbtronics, 2024-06-22:
> 2\. Some string translations are missing when using the en_DE locale. So far I only noticed the search placeholder "search.placeholder" and the indicator for adding a new catecory/footprint/etc. "entity.select.group.not_added_to_DB"

This is now fixed with the latest commit

suuppl, 2024-10-11:
I think it would be a very nice option to pass a custom datetime formatter to IntlDatetimeFormatter.create() ([`LocaleDateTimeColumn.php:L61`](https://github.com/Part-DB/Part-DB-server/blob/58ed57fab7dcdbe2a60fb872e0f38394148e27fd/src/DataTables/Column/LocaleDateTimeColumn.php#L61)), using the optional `pattern`, as described in the [docs](https://www.php.net/manual/en/intldateformatter.create.php). 

I don't understand where the options (`$this->options`) are set, but i guess it's possible to have an optional `customDateTimeFormat` field in the options (example ISO8601-ish: `yyyy-MM-dd HH:mm:ss`, [formatting docs](https://unicode-org.github.io/icu/userguide/format_parse/datetime/#datetime-format-syntax)) and pass an appropriate `pattern`? (I also don't really know php or I'd try it myself)