Part-DB/Part-DB-server #568: Error during creation: Please check your inputs!

AlexD0302, 2024-03-13:
Hello, 
I have sometimes the error that when I want to edit an existing component the 
Error message: "Error during creation: Please check your inputs!" appears. 

If I delete all "incorrect" fields, I jump back to the "Common" tab and it is still not possible to save. the changes

Does anyone have the same problem or am I doing something wrong?

jbtronics, 2024-03-16:
What are the fields which are invalid and what message does they show?

AlexD0302, 2024-03-16:
I mean the following fields (this is just a example and many more Parts have this strange behavior)

[image]

Even if i delete the marked fields, the error will appear again and shows that now an other field isn't correct.

eleinfroot, 2024-03-29:
Sorry, my english is poor :)
When i update part from information provider, saving changes is OK, but if i try to change anything after, then i got same errors as above. Workaround, not solution: delete all parameters, purchase informations and attachments and save changes. Then again update from info providers and save.

jbtronics, 2024-03-29:
You have to delete the duplicate parameters (the ones marked red), before saving. Then the duplicate error message should disappear

jbtronics, 2024-03-29:
What part provider do You use? And what part is it?

I guess the provider returns some duplicate fields (which is not allowed).

eleinfroot, 2024-04-02:
Yes. TME returns duplicate fields like "rodzaj opakowania", but first save is OK. Only after changing already imported information does the error appear.
[image]
[image]
[image]

d-buchmann, 2024-04-11:
Maybe it's possible - instead of filling in the duplicate parameters on part review - to add just the first parameter and join the duplicates as a comma separated list into the text field of the first one? I think little to no information will be lost that way, and ultimately the info provider (TME) is at fault here.

jbtronics, 2024-04-28:
The validation is now performed already on part creation, meaning you can not create an invalid part (a message will be shown then). The user can then decide which value to keep.

closed 2024-04-28 by jbtronics