Part-DB/Part-DB-server #572: Take photos

klobs, 2024-03-21:
**Is your feature request related to a problem? Please describe.**
While inventorizing my stock parts and tools, would like to directly take a picture of the items directly with my mobile phone and use them as an attachment. I want to avoid having to switch apps for photo taking and storing photos locally.

**Describe the solution you'd like**
From within the part edition dialogue, esp. the attachment subsection, I would be able to have the possibility to take a picture (with my mobile). Besides URL and upload, provide a third button that allows me to take a picture.

**Describe alternatives you've considered**
Take a photo, store it locally, and then from within the file system dialog pick identify the picture, select it, upload it and consequently delete it.

SANCLA, 2024-03-22:
I also support this request, seems immensely useful.
In addition, I am also willing to pay if this can be offered in app form.

Just an addition, though, that modern phones take very high-resolution photos of considerable size. 
Perhaps also a great opportunity to automatically resize photos to a more efficient and compact size.

PC-maniak, 2024-12-27:
+1 definitely =)