Part-DB/Part-DB-server #579: Part-DB Installation causes server to freeze and unresponsive

philm001, 2024-03-29:
**Describe the bug**

I am installing part-DB on a AWS EC2 instance and while installing, my EC2 isntance becomes unresponsive and I am unable to SSH into it. A reboot of the instance does not fix the problem. This happens on both the docker installation and on the manual installation.

For the docker, after I execute this command:

`docker-compose up -d`

after about 10-15 sec, the instance freezes and even after reboot, I am unable to SSH into it.

For the manual installation, after executing these commands, does the instance become unstable and eventually, it too is un responsive:

`# Install composer dependencies (please note the sudo command, to run it under the web server user)
sudo -u www-data composer install --no-dev -o

# Install yarn dependencies
sudo yarn install
# Build frontend
sudo yarn build`

I should note that I am using a MySQL database and not the SQLLite.

I am not sure what log information I can get that will assist in figuring this out. Please let me know and I can see if I can get it for you.

I would prefer to use MySQL since I will have various users accessing the part-DB server.

**To Reproduce**
Steps to reproduce the behavior:

Follow the instructions on installing the Part-DB server along with the instructions on installing MySQL (3rd party instructions will be needed to setup the server)

**Expected behavior**
EC2 instance does not freeze up

**Screenshots**
N/A

**Server Side**
- Part-DB Version: latest stable release
- PHP Version: PHP-8.3
- Database Server: MySQL latest version as of 3/29/2024

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser N/A
 - Version 22.04

**Smartphone (please complete the following information):**
N/A

**Additional context**
N/A

jbtronics, 2024-03-29:
I think that is a problem with your server (or it's configuration), not a bug in Part-DB. Normally a user space application should not be able to destabilize a system.

Maybe your server runs out of RAM or CPU time because especially the `yarn build` step is pretty CPU and RAM intensive.

philm001, 2024-03-29:
I see, the EC2 instance is using 1 GB of RAM and the CPU core is 1 CPU. Is there a recommended spec for the server to run part-DB?

jbtronics, 2024-03-29:
I would say, that this should be enough to run Part-DB (even though I don't understand why docker causes problems). But maybe not enough to compile the frontend assets.

However instead of doing the steps with composer and yarn yourself, you can just copy one of the ZIP files from the release pages, which contains pre built assets.

philm001, 2024-03-29:
Ok, I was following the steps in the guide that was provided. Should I unzip it in the folder `/var/www/partdb`?

jbtronics, 2024-03-29:
If that is where Your part-DB should live, then yes

philm001, 2024-03-29:
Ok, so basically, I am going to follow the steps outlined here: https://docs.part-db.de/installation/installation_guide-debian.html

Except, I am going to use the pre-built asset/dependencies pack. Do I still need to install PHP and the like? Apache will still need to be installed.

As for hte MySQL server, I take it that I still need to configure it accordingly to use the MySQL database? I take it that the DATABASE_URL still needs to be set in the .env.local file?

jbtronics, 2024-03-29:
Yes, you need to do everything else. The pre built assets just replaces the composer and yarn command steps. And maybe the git clone step if you use the "Full Part-DB" zip file.

And you still need to configure the database and everything else like normal.

philm001, 2024-04-01:
Hello, still working on the setup part before launching everything but so far things are working out. 

Quick question, do you have any guides/documentation on how to setup the Digikey provider for Part-DB?

philm001, 2024-04-03:
Great news, we ended up changing directions. Instead of running an EC2 instance, we found a way to use local hardware and still have access to it on the internet. I was able to get things up and running with the docker container.

One final question, Digikey is asking for a callback for the oauth. Where do I point the callback? Is it to the part DB server?

philm001, 2024-04-04:
@jbtronics

jbtronics, 2024-04-04:
> One final question, Digikey is asking for a callback for the oauth. Where do I point the callback? Is it to the part DB server?

I think this setting can be left empty on the digikey page

philm001, 2024-04-04:
Ok, I see, I got it.

So what happened is that I enabled v4 for the product information and not v3.

Currently on Digikey, v3 is set as depreciated and they are using v4. 

I would recommend to include some info on the installation guide for people to use v3 and not v4.

I will also open another issue to update the API in Part-DB to use Digikey v4

@jbtronics

closed 2024-04-04 by philm001