Part-DB/Part-DB-server #587: Footprint autocomplete will suggest HTML elements

Gnarflord, 2024-04-06:
**Describe the bug**
When I begin typing in the field for my KiCad footprint I will get HTML elements as suggestions:
[image]
(Resistor_R:R_1206_3216Metric was added manually by me, I don't get any auto-completes from the default library)

Now If I empty the field I can see the whole HTML document being parsed. This is a fun way of debugging I've never encountered before :)
[image]

[image]

Now that's basically my Apache server telling me that there's a 404 error and the URL does not exist. If I look into the Apache log I see:

```
[06/Apr/2024:11:00:19 +0200] "GET /kicad/footprints.txt HTTP/1.1" 404 1137
```
Aha! I'm hosting under a subpath like `https://myurl.com/partdb` but the footprint lookup searches under `https://myurl.com` without the subpath. My Apache config for the reverse proxy looks like this:

```
<Location "/partdb">
        ProxyPreserveHost On
        ProxyPass "http://localhost:8081"
        ProxyPassReverse "http://localhost:8081"
        RequestHeader set X-Forwarded-Prefix "/partdb"
</Location>
```

Maybe I've set something up the wrong way but `X-Forwarded-Prefix` is definitely working as the page breaks completely if it's missing. But somehow `X-Forwarded-Prefix` is ignored for this particularly footprint lookup.

I have also set `DEFAULT_URI` to `https://myurl.com/partdb/` but that doesn't seem to have any impact.

**Server Side**
- Part-DB Version: 1.11.3 (via docker)
- Apache: 2.4.58-1

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox

Gnarflord, 2024-04-08:
I have also noticed that the URLs for API access are not correct, which seems to be a related issue:

[image]

### What I expect
```
Part-DB API: https://myurl.com/partdb/api
KiCad API root URL: https://myurl.com/partdb/en/kicad-api/
```
### What I got
```
Part-DB API: http://myurl.com/partdb/api
KiCad API root URL: http://myurl.com/en/kicad-api/
```

So the issues are:
- http instead of https
- KiCad API is missing the subpath `partdb` which the Part-DB API got correct

jbtronics, 2024-04-10:
Have you set the correct TRUSTED_PROXIES env for Part-DB?

Gnarflord, 2024-04-10:
My env contains: `TRUSTED_PROXIES=127.0.0.1,REMOTE_ADDR`
As to my understanding that basically allows connections from any address, right?

jbtronics, 2024-04-28:
This is now fixed

closed 2024-04-28 by jbtronics