Part-DB/Part-DB-server #590: LDAP Integration

Jedi1714, 2024-04-11:
Hi,
I don't have clear how I can configure correctly my Part-DB using LDAP users.
Do you have any specific protocol or documentation for that?

Thanks an advance

Daniel

jbtronics, 2024-04-11:
Part-DB can not use LDAP directly (as LDAP is not really a good protocol for web application). However Part-DB supports single sign on via SAML.

If you setup a SAMl identity provider like keycloak you can configure it to use LDAP as backend. Part-DB can then use keycloak as identity provider to allow authentication of users from an LDAP. 

https://docs.part-db.de/installation/saml_sso.html

Jedi1714, 2024-04-12:
Hi,

OK, Rhanks for your fast answer.
I try to configure using SAML

closed 2024-04-18 by jbtronics