Part-DB/Part-DB-server #598: Add SI unit prefixes in parts parameters

mdiepart, 2024-04-17:
This is a feature request that should allow for better comparisons and research among the components in the database.
Let's take capacitors as an example. A 100nF capacitor could be registered as a 0.1 uF capacitor or a 100 nF capacitor. The exact value is the same but the representation is different. Thus, when saving the capacitance as a parameter, we could have : value 0.1, unit uF or value 100 unit nF. The goal of this feature would be to support SI unit prefixes such that we would enter either : value 0.1u unit F or value 100n unit F. Both would give the same result once saved in the database (saved as value 0.0000001 unit F).

One possibility is to add a checkbox next to the unit field. When ticked, it would mean that the unit supports SI prefixes and would enable the possibility to use SI prefixes in the various value fields. 

The current alternative would be to either express all the capacitors as a multiple of the smallest value possible (i.e. store all capacitors as multiples of fF). This would quickly become impractical especially when dealing with i.e. 1000000000 fF for 1 uF.

I tried to look into the code but I'm not fluent at all in PHP and the symfony framework. I can help if needed or I can try to implement a first version if given explicit steps of what to do.

tobiasfalk, 2025-03-08:
One other posibilety would be to enable Parameters to have a Measurment Unit, since there we already have a System to handle SI prefixes

tobiasfalk, 2025-03-08:
Also a "compromis" for start would be if numbers could be displayed in Sientidic notation, this way we can avoid 10k being displayed as 10000. I know it already acapts it and when reaching a thrashold it already displays it in sienific notation, which sound to me as if this threashold cold be lowerd or made a option for the user.

Additionaly if the e^could be in multiples of 3 it would be ideal