Part-DB/Part-DB-server #599: Sync over multiple instances

bajticzek, 2024-04-17:
Hi, foremost thanks for your hard work, this app is amazing. I discovered it today as 1000+ parts list in Excel is becoming frustrating to use. This is perfect solution.

To my problem:
I installed 2 instances in Docker. One is on my work PC, the other at home. Some parts I have at home, some at work, so I need to use the DB on both machines. Neither can be open to outside world, so that's why I need to local instances. Is there some sort of script or automated batch file that I can run so the DB is saved to one file which I can merge into the DB at the other location? I am looking for a simple solution (use-wise). If I need to take many preliminary steps that results in a single click file, that's OK. Just so I can before going/after leaving the work have my DB with me.

Thanks for any insight.

frank-f, 2024-04-18:
Why don't you just run one Part-Db instance that can be reached from both locations?

jbtronics, 2024-04-23:
I agree that, the best solution would be to install Part-DB on a server in the internet, which is available from all location you wanna use it from.

Permanently keeping two instances in sync, without running into inconsistencies is very difficult to impossible and would probably require heavy changes to Part-DB internal architecture. Something like this is not planned.

closed 2024-04-23 by jbtronics