Part-DB/Part-DB-server #603: Cost calculation after building project

MG-5, 2024-04-21:
First of all, your project is really great, practical and I'm glad that it makes my everyday assembly so much easier. 

It would be cool if, as soon as you build a project, the total price of all items for which a unit price exists could be calculated and displayed.

partywagon112, 2024-05-14:
Had the same thought when I imported my project. Looks like doing something like that is supported inthe API, but definitely would be amazing with a native indication price based on the item data.

mase76, 2024-06-30:
That would be a great feature. With that you could offer buildkits in your Hackerspace.

chickey, 2024-12-18:
Just wanted to bump this up as i think it would be very useful and something i liked from parts box.  I don't mind it being worst case pricing without accommodating price breaks etc.  I'm a small maker so rarely have multiples of the same part in from different manufacture's so that would not be an issue either.  Partsbox also allowed you to input the quantity to get a rough bom cost.

[image]

JochenVandorpe, 2025-10-25:
+1 (for now still tracking this in excel based on the schematic BOM but feels like overhead when creating the projects in part-DB to 'book' them there for parts count update)