Part-DB/Part-DB-server #604: About kicad‘s boom import Part-DB

QM1412, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
I want to use Part-DB to manage the project, but I got an error when importing the boom table of kicad, and I am using kicad version 8.0

**Describe the solution you'd like**
I would like to know how to properly import boom tables to Part-DB

**Additional context**
Add any other context or screenshots about the feature request here.
this is my boom table with kicad 8.0
[image]
this is my error ,if i import the boom table ,it will report this error
[image]
i don't know how to modify it

jbtronics, 2024-04-28:
This was caused by the chinese header files (Part-DB expected english or german ones). Now the importer is language invariant, as long as KiCad does not change the ordering of the files.

closed 2024-04-28 by jbtronics

QM1412, 2024-05-02:
**I've changed the language format of the boom table, but I get the following error, what's going on**

[image]
[image]