Part-DB/Part-DB-server #608: PartKeepr to Part-DB migration in docker

pawlef, 2024-05-02:
Hi all,
I would like to move from the discontinued Partkeeper to the promising Part-DB, but errors appear during import.
According to the instructions, I exported the database from Partkeepr to an xml file.
I placed the file in the directory and executed the command
docker exec --user=www-data -it partdb bin/console partdb:migrations:import-partkeepr /var/www/html/var/db/pk.xml

I tried the latest and edge branch, but unfortunately without success.
I am attaching my pk.xml file and a screenshot.

docker exec --user=www-data -it partdb bin/console partdb:migrations:import-partkeepr /var/www/html/var/db/pk.xml

 ! [NOTE] This command is still in development. If you encounter any problems, please report them to the issue tracker  
 !        on GitHub.                                                                                                    

                                                                                                                        
 [WARNING] This command will delete all existing data in the database (except users). Make sure that you have no        
           important data in the database before you continue!                                                          
                                                                                                                        

 Please type "DELETE ALL DATA" to continue. []:
 > DELETE ALL DATA

                                                                                                                        
 [INFO] Importing distributors...                                                                                       
                                                                                                                        

                                                                                                                        
 [OK] Imported 5 distributors.                                                                                          
                                                                                                                        

                                                                                                                        
 [INFO] Importing part measurement units...                                                                             
                                                                                                                        

                                                                                                                        
 [OK] Imported 1 measurement units.                                                                                     
                                                                                                                        

                                                                                                                        
 [INFO] Importing manufacturers...                                                                                      
                                                                                                                        

                                                                                                                        
 [OK] Imported 288 manufacturers.                                                                                       
                                                                                                                        

                                                                                                                        
 [INFO] Importing categories...                                                                                         
                                                                                                                        

                                                                                                                        
 [OK] Imported 272 categories.                                                                                          
                                                                                                                        

                                                                                                                        
 [INFO] Importing Footprints...                                                                                         
                                                                                                                        

                                                                                                                        
 [OK] Imported 44 footprints.                                                                                           
                                                                                                                        

                                                                                                                        
 [INFO] Importing storage locations...                                                                                  
                                                                                                                        

                                                                                                                        
 [OK] Imported 32 storage locations.                                                                                    
                                                                                                                        

                                                                                                                        
[pk.tgz.txt](https://github.com/Part-DB/Part-DB-server/files/15190217/pk.tgz.txt)
       
                                                                                                                        

{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211","trace":["/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:102","/var/www/html/vendor/doctrine/dbal/src/Connection.php:387","/var/www/html/vendor/doctrine/dbal/src/Connection.php:453","/var/www/html/vendor/doctrine/dbal/src/Connection.php:411","/var/www/html/vendor/doctrine/dbal/src/Connection.php:318","/var/www/html/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:126","/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:46","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:82","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1096","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:126","/var/www/html/vendor/symfony/console/Application.php:324","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:175","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:49","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:18"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2024-05-02T14:54:36.877203+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/var/www/html/var/db/pk.xml"}}}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener [... cut ...]
In Exception.php line 28:
                                                                                           
  SQLSTATE[23000]: Integrity constraint violation: 19 UNIQUE constraint failed: parts.ipn  
                                                                                           

In Statement.php line 130:
                                                                                           
  SQLSTATE[23000]: Integrity constraint violation: 19 UNIQUE constraint failed: parts.ipn

jbtronics, 2024-05-02:
You probably have two different parts in your partkeepr database which have the same internal part number (IPN). For partkeepr that does not matter, but for Part-DB that is intended as a custom internal identifier for this specific part and it must be unique across the database.

So you should fix that in your data before proceeding and change (or remove) the IPN field of one of the parts before continuing with the import to Part-DB.

Either you do the changes in PartKeepr or you do it in the XML file directly.

pawlef, 2024-05-02:
Thank you for your very quick help. I found and removed duplicates in the xml file.

closed 2024-05-02 by pawlef