Part-DB/Part-DB-server #617: Can't accessing detail pages

Katjabo, 2024-05-23:
i have upgraded my local installation from 0.5.15 (i guess) to the newest one.
Now i have trouble to accessing detail pages for the parts or projects.
tools works fine.

**To Reproduce**
get the latest Part-DB
install latest composer, Yarn and Xampp
Install local (xampp independing mysql server ( i use 8)
Have an exsiting DB from Parts-DB Version 0.5.15

Steps to reproduce the behavior:
extract the Part-DB from Zip into httpdocs folder of xampp
Follow the installation Guide of Part-DB from point 1 until 7

try it out.

**Expected behavior**
inspectable parts and projects on webpage

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.12
- PHP Version: PHP 8.2.12
- Database Server mysql 8.0.33

**Desktop (please complete the following information):**
 - OS: windows
 - Browser firefox
 - Version 126.0

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Log entries:

[2024-05-23T12:03:38.516499+02:00] deprecation.INFO: User Deprecated: Since symfony/doctrine-bridge 6.3: Registering "App\EventSubscriber\LogSystem\EventLoggerSubscriber" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. at C:\\xampp_8\\htdocs\\PartDB\\vendor\\symfony\\doctrine-bridge\\ContainerAwareEventManager.php:211)"} {"token":null,"url":"/PartDB/public/en/category/2/parts","ip":"127.0.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/PartDB/public/en/category/38/parts"}
[2024-05-23T12:03:38.521449+02:00] deprecation.INFO: User Deprecated: Since symfony/doctrine-bridge 6.3: Registering "App\EventSubscriber\LogSystem\LogDBMigrationSubscriber" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. at C:\\xampp_8\\htdocs\\PartDB\\vendor\\symfony\\doctrine-bridge\\ContainerAwareEventManager.php:211)"} {"token":null,"url":"/PartDB/public/en/category/2/parts","ip":"127.0.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/PartDB/public/en/category/38/parts"}
[2024-05-23T12:03:39.246107+02:00] deprecation.INFO: User Deprecated: Since symfony/doctrine-bridge 6.3: Registering "App\EventSubscriber\LogSystem\EventLoggerSubscriber" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. at C:\\xampp_8\\htdocs\\PartDB\\vendor\\symfony\\doctrine-bridge\\ContainerAwareEventManager.php:211)"} {"token":null,"url":"/PartDB/public/PartDB/public/en/category/2/parts","ip":"127.0.0.1","http_method":"POST","server":"localhost","referrer":"https://localhost/PartDB/public/en/category/2/parts"}
[2024-05-23T12:03:39.249178+02:00] deprecation.INFO: User Deprecated: Since symfony/doctrine-bridge 6.3: Registering "App\EventSubscriber\LogSystem\LogDBMigrationSubscriber" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute. at C:\\xampp_8\\htdocs\\PartDB\\vendor\\symfony\\doctrine-bridge\\ContainerAwareEventManager.php:211)"} {"token":null,"url":"/PartDB/public/PartDB/public/en/category/2/parts","ip":"127.0.0.1","http_method":"POST","server":"localhost","referrer":"https://localhost/PartDB/public/en/category/2/parts"}

jbtronics, 2024-05-23:
The paths are wrongly generated. 
Putting Part-DB into a subfolder is not recommended, as its really difficult to setup, especially in a constellation like yours.

You should setup a vhost where the DocumentRoot is pointing to the `public/` folder of Part-DB.

Katjabo, 2024-05-23:
i develop some web Apps thats why i use xampp. I dont want to setup up a vhost, so i'll downgrade the part db to 0.5
wich is working nice under xampp

Katjabo, 2024-05-23:
x

closed 2024-05-23 by Katjabo