Part-DB/Part-DB-server #618: Export and print functions for all tables

RaptorDE, 2024-05-23:
It would be nice if you could export tables as they are displayed.

One application would be to export the BOM of a project.

A simple CSV export would be enough, the possibility to print directly or create a PDF would be a plus.

As a workaround, you can copy and paste the table or create a screenshot. However, this is not so nice for further data usage in a spreadsheet.

Datinhkhach, 2024-08-12:
I have used the Snipe-it export feature and it is great. The data is exactly as it is shown. The export file format is also versatile and useful, I often use it to export to excel for editing.

[image]

Donatussss, 2025-07-01:
Hello, how can I do this, I'm a bit new to part-db. I will appreciate your help, thank you

Donatussss, 2025-07-16:
@RaptorDE , ohh, I meant to say how to I export a csv of all the parts in part-db