Part-DB/Part-DB-server #632: Strange behavior at importing BOM in project

MG-5, 2024-06-18:
**Describe the bug**
Import BOM at project creation will throw error 500.
Delete cache and restart docker container doesnt help.

***Workaround:*** First create project (without importing) and then go to "BOM" -> "Import BOM" works

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Edit->Projects->New Project->Import/Export
2. Try to load KiCad BOM
3. See error 500

**Server Side**
- Part-DB Version: 1.12.1 Docker

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu Budgie 24.04
 - Browser : Firefox 127

**Additional context**
docker logs:

```
- -  18/Jun/2024:10:02:19 +0000 &quot;GET /index.php&quot; 200
172.22.0.5 - - [18/Jun/2024:10:02:19 +0000] &quot;GET /de/project/new HTTP/1.1&quot; 200 13966 &quot;https://partdb.[replaced_domain].de/de/project/new&quot; &quot;Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:127.0) Gecko/20100101 Firefox/127.0&quot;
{&quot;message&quot;:&quot;Matched route \&quot;project_new\&quot;.&quot;,&quot;context&quot;:{&quot;route&quot;:&quot;project_new&quot;,&quot;route_parameters&quot;:{&quot;_route&quot;:&quot;project_new&quot;,&quot;_controller&quot;:&quot;App\\Controller\\AdminPages\\ProjectAdminController::new&quot;,&quot;_locale&quot;:&quot;de&quot;},&quot;request_uri&quot;:&quot;http://partdb.[replaced_domain].de/de/project/new&quot;,&quot;method&quot;:&quot;POST&quot;},&quot;level&quot;:200,&quot;level_name&quot;:&quot;INFO&quot;,&quot;channel&quot;:&quot;request&quot;,&quot;datetime&quot;:&quot;2024-06-18T10:02:45.718825+00:00&quot;,&quot;extra&quot;:{&quot;token&quot;:null,&quot;url&quot;:&quot;/de/project/new&quot;,&quot;ip&quot;:&quot;172.22.0.5&quot;,&quot;http_method&quot;:&quot;POST&quot;,&quot;server&quot;:&quot;partdb.[replaced_domain].de&quot;,&quot;referrer&quot;:&quot;https://partdb.[replaced_domain].de/de/project/new&quot;}}
{&quot;message&quot;:&quot;Read existing security token from the session.&quot;,&quot;context&quot;:{&quot;key&quot;:&quot;_security_main&quot;,&quot;token_class&quot;:&quot;Symfony\\Component\\Security\\Core\\Authentication\\Token\\UsernamePasswordToken&quot;},&quot;level&quot;:100,&quot;level_name&quot;:&quot;DEBUG&quot;,&quot;channel&quot;:&quot;security&quot;,&quot;datetime&quot;:&quot;2024-06-18T10:02:45.722432+00:00&quot;,&quot;extra&quot;:{&quot;token&quot;:null,&quot;url&quot;:&quot;/de/project/new&quot;,&quot;ip&quot;:&quot;172.22.0.5&quot;,&quot;http_method&quot;:&quot;POST&quot;,&quot;server&quot;:&quot;partdb.[replaced_domain].de&quot;,&quot;referrer&quot;:&quot;https://partdb.[replaced_domain].de/de/project/new&quot;}}
{&quot;message&quot;:&quot;User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \&quot;App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\&quot; as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.&quot;,&quot;context&quot;:{&quot;exception&quot;:{&quot;class&quot;:&quot;ErrorException&quot;,&quot;message&quot;:&quot;User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \&quot;App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\&quot; as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.&quot;,&quot;code&quot;:0,&quot;file&quot;:&quot;/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211&quot;,&quot;trace&quot;:[&quot;/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:102&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:387&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:453&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:411&quot;,&quot;/var/www/html/vendor/doctrine/dbal/sr
c/Connection.php:318&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/Persisters/Entity/BasicEntityPersister.php:211&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/UnitOfWork.php:3514&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/EntityManager.php:504&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/EntityRepository.php:197&quot;,&quot;/var/www/html/vendor/symfony/doctrine-bridge/Security/User/EntityUserProvider.php:92&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/ContextListener.php:207&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/ContextListener.php:127&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/AbstractListener.php:26&quot;,&quot;/var/www/html/vendor/symfony/security-bundle/Security/LazyFirewallContext.php:60&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall.php:128&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall.php:95&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:260&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:220&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/EventDisp
atcher.php:56&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:157&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/Kernel.php:197&quot;,&quot;/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35&quot;,&quot;/var/www/html/vendor/autoload_runtime.php:29&quot;,&quot;/var/www/html/public/index.php:5&quot;]}},&quot;level&quot;:200,&quot;level_name&quot;:&quot;INFO&quot;,&quot;channel&quot;:&quot;deprecation&quot;,&quot;datetime&quot;:&quot;2024-06-18T10:02:45.729189+00:00&quot;,&quot;extra&quot;:{&quot;token&quot;:null,&quot;url&quot;:&quot;/de/project/new&quot;,&quot;ip&quot;:&quot;172.22.0.5&quot;,&quot;http_method&quot;:&quot;POST&quot;,&quot;server&quot;:&quot;partdb.[replaced_domain].de&quot;,&quot;referrer&quot;:&quot;https://partdb.[replaced_domain].de/de/project/new&quot;}}
{&quot;message&quot;:&quot;User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \&quot;App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\&quot; as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.&quot;,&quot;context&quot;:{&quot;exception&quot;:{&quot;class&quot;:&quot;ErrorException&quot;,&quot;message&quot;:&quot;User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \&quot;App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\&quot; as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] or #[AsDocumentListener] attribute.&quot;,&quot;code&quot;:0,&quot;file&quot;:&quot;/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211&quot;,&quot;trace&quot;:[&quot;/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:102&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:387&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:453&quot;,&quot;/var/www/html/vendor/doctrine/dbal/src/Connection.php:411&quot;,&quot;/var/www/html/vendor/doctrine/d
bal/src/Connection.php:318&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/Persisters/Entity/BasicEntityPersister.php:211&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/UnitOfWork.php:3514&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/EntityManager.php:504&quot;,&quot;/var/www/html/vendor/doctrine/orm/src/EntityRepository.php:197&quot;,&quot;/var/www/html/vendor/symfony/doctrine-bridge/Security/User/EntityUserProvider.php:92&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/ContextListener.php:207&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/ContextListener.php:127&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall/AbstractListener.php:26&quot;,&quot;/var/www/html/vendor/symfony/security-bundle/Security/LazyFirewallContext.php:60&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall.php:128&quot;,&quot;/var/www/html/vendor/symfony/security-http/Firewall.php:95&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:260&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:220&quot;,&quot;/var/www/html/vendor/symfony/event-dispatcher/Eve
ntDispatcher.php:56&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:157&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76&quot;,&quot;/var/www/html/vendor/symfony/http-kernel/Kernel.php:197&quot;,&quot;/var/www/html/vendor/symfo [... cut ...] uot;http_method&quot;:&quot;POST&quot;,&quot;server&quot;:&quot;partdb.[replaced_domain].de&quot;,&quot;referrer&quot;:&quot;https://partdb.[replaced_domain].de/de/project/new&quot;}}
172.22.0.5 - - [18/Jun/2024:10:02:45 +0000] &quot;POST /de/project/new HTTP/1.1&quot; 500 4273 &quot;https://partdb.[replaced_domain].de/de/project/new&quot; &quot;Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:127.0) Gecko/20100101 Firefox/127.0&quot;
- -  18/Jun/2024:10:02:45 +0000 &quot;POST /index.php&quot; 500
```

jbtronics, 2024-06-22:
The import/export tab is not for importing BOM lists, but to import whole projects, which were exported by Part-DB before (like it does for every other datastructure).

To import BOM tables from KiCAD you have to use the BOM Import feature, under the BOM tab. It is only availble for already created projects, therfore this button is not visible during project creation.

jbtronics, 2024-06-22:
You should not encounter a 500 error however, but this is now fixed.

closed 2024-06-22 by jbtronics