Part-DB/Part-DB-server #633: Backup detects SQLite while our installation uses MySQL

au-ee, 2024-06-18:
**Describe the bug**
We have 1.12.0 installed on Kubernetes. Everything is configured through environment variables. Excerpt from our deployment's yaml:

```yaml
        - name: DATABASE_URL
          valueFrom:
            secretKeyRef:
              name: partdb-database-access
              key: db-url # this field holds something like mysql://partdb:PASSWORD@partdb-database:3306/partdb
        - name: APP_ENV
          value: docker
        - name: DEFAULT_LANG
          value: en
        - name: DEFAULT_TIMEZONE
          value: Europe/Berlin
        - name: BASE_CURRENCY
          value: EUR
        - name: INSTANCE_NAME
          value: Part-DB
        - name: ALLOW_ATTACHMENT_DOWNLOADS
          value: "1"
        - name: USE_GRAVATAR
          value: "1"
        - name: BANNER
          value: "Part-DB hosted on BBN K8S"
        - name: PROVIDER_DIGIKEY_CLIENT_ID
          valueFrom:
            secretKeyRef:
              name: partdb-provider-access
              key: PROVIDER_DIGIKEY_CLIENT_ID
        - name: PROVIDER_DIGIKEY_SECRET
          valueFrom:
            secretKeyRef:
              name: partdb-provider-access
              key: PROVIDER_DIGIKEY_SECRET
```

Part-DB works and uses its MySQL db. The backup thinks we have SQLite.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the Part-DB pod: `kubectl exec -it pod-name -- bash`
2. Execute backup: `sudo -u www-data php bin/console partdb:backup --database /tmp/partdb.zip` which results in the following output:

```
 [INFO] Backup Part-DB to /tmp/partdb.zip

 ! [NOTE] Starting backup...

 ! [NOTE] Backup database...

 ! [NOTE] SQLite database detected. Copy DB file to ZIP...

 [OK] Backup finished! You can find the backup file at /tmp/partdb.zip
```

The db file contains the empty file `app.db` from `/var/www/html/var` which is part of the container image.

**Expected behavior**
That the backup system detects that part-db is powered by a MySQL db and uses mysqldump to create the backup.

**Server Side**
- image: jbtronics/part-db1:v1.12.0

**Desktop (please complete the following information):**
n/a

**Smartphone (please complete the following information):**
n/a

jbtronics, 2024-06-19:
What does the command `bin\console doctrine:migrations:status` show? And what does the server info page says in the database tab?

If you start a console in the container, does it has access to the secret envs ?

Part-DB does the decision, what to do on the database platform the connection does. If that would be wrong, you would probably encounter a lot of other issues.

au-ee, 2024-06-20:
While collecting information, I discovered that I did not specify `-E` in my `sudo`, so the env variables where missing for `www-data`.

This way, it works:

`sudo -u www-data php -E bin/console partdb:backup --database /tmp/partdb.zip`

Problem solved. Thanks for the hints!

closed 2024-06-20 by au-ee