Part-DB/Part-DB-server #636: Info providers - cannot connect with anyone

Klewciax, 2024-06-23:
Hi, I have problem with configure info providers. I prepare configurations with tokens in service.yaml but I still have disabled TME, MOUSER, LSCS what i configured. I also change the '%env' parameters but without result could anyone help me with this configuration ?

Klewciax, 2024-06-23:
Maybe it would be helpful but if I add env(TME_API_keys) lines I've got a info :  There is no extension able to load the configuration for "env(PROVIDER_TME_KEY)" (in "/var/www/partdb/config/services.yaml"). Looked for namespace "env(PROVIDER_TME_KEY)", found ""fram
  ework", "doctrine", "doctrine_migrations", "security", "twig", "monolog", "webpack_encore", "datatables", "shivas_versioning", "liip_imagine", "twig_extra", "gregwar_captcha", "transla
  tion", "florianv_swap", "nelmio_security", "turbo", "tfa_webauthn", "scheb_two_factor", "webauthn", "nbgrp_onelogin_saml", "stimulus", "ux_translator", "dompdf_font_loader", "knpu_oaut
  h2_client", "nelmio_cors", "api_platform""

jbtronics, 2024-06-23:
You do not need to change anything in the services.yaml file 

You configure the info providers like anything else in you `.env.local` file. If you copied it from the `.env` file recently, you will find there already some lines for the info provider configuration (otherwise you need to copy them from .env)

Klewciax, 2024-06-23:
My current configuration on services.yaml: 
```
414 env(PROVIDER_TME_KEY): 'mykey'
415 env(PROVIDER_TME_SECRET): 'mykey'
416 env(PROVIDER_TME_CURRENCY): 'PLN'
417 env(PROVIDER_TME_COUNTRY): 'PL'
418 env(PROVIDER_TME_LANGUAGE): 'pl'
419 env(PROVIDER_TME_GET_GROSS_PRICES): 'true'
420
421 env(PROVIDER_MOUSER_KEY): 'mykey'
422 env(PROVIDER_MOUSER_SEARCH_WITH_SIGNUP_LANGUAGE):
423 env(PROVIDER_MOUSER_SEARCH_OPTION): 'RohsAndInStock'
424 env(PROVIDER_MOUSER_SEARCH_LIMIT): '50'

```
Or where can I find .env.local ?

jbtronics, 2024-06-23:
Do not put these in services.yaml. the values there are just a fallback, when nothing else is defined. As a `.env` file exist, the values in services.yaml gets ignored.

The `.env.local` file should exist at the part-DB main folder. If not duplicate the `.env` file and rename it `.env.local`.

See the installation docs: https://docs.part-db.de/installation/installation_guide-debian.html

Under the "Create configuration for Part-DB".

Klewciax, 2024-06-23:
After create .env.local everything works fine :)

closed 2024-06-23 by Klewciax