Part-DB/Part-DB-server #637: Usage of API token will logout the web frontend

whc2001, 2024-06-23:
**Describe the bug**
When using the WebUI and the API token from the same user at the same time, the request with API token will invalidate the WebUI's current login state.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an API token from the current user
2. Access the API interactive documentation page and authenticate with the generated API token
3. Validate that WebUI is working fine
4. Send any request with API interactive documentation page (here I tried GET `/api/parts`)
5. Go back to WebUI and do any navigation (even clicking the top left link to home page), will be redirected to login page with warning `Access denied! Please log in to continue.`

**Expected behavior**
WebUI and API token access should be simultaneous without mutual exclusion.

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.12.1
- PHP Version: 8.2.18 (SAPI: fpm-fcgi)
- Database Server 10.11.6-MariaDB-0+deb12u1

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Microsoft Edge
 - Version: 125.0.2535.92

**Additional context**
This could be circumvented by creating another user, create an API token there then set an unreasonably long password so that it's basically unloginable manually (`User disabled (no login possible)` cannot be used since it will also disable API token access). However I still think this might be a bug.

jbtronics, 2024-06-23:
I dont think there is an easy fix for this, as we need the session, and the symfony authentication mechanism resets the session every time when a new authentication occurs. (And sending an API request with the API token is such a authentication).

But I am not really sure if that it is really a problem. If you want to try out the API endpoints using the web ui, you do not need the API token authentications. They will respond just fine, even without the token set, as they use your existing authentication.

And in API applications, where you need the API token for authentication, you have a seperate session context, where using the API token authentication is not affecting the WebUI in the browser.

whc2001, 2024-06-23:
Ahh sorry and nevermind, seems like it's the interactive documentation causing the problem. I have tried again with `curl` and seems like the problem went away.

closed 2024-06-23 by whc2001