Part-DB/Part-DB-server #639: Search by ID

jon-daemon, 2024-06-25:
I have printed labels for all my parts which contain the name and the ID.
It would be really helpful to be able to use ID in the search bar. Is it something that could be implemented?

d-buchmann, 2024-06-25:
There is no need to "search" for an ID because it is unique within Part-DB and thus will always yield a single result.
You can directly access any part by typing `<your.partdb.instance>/part/<ID>` into your address bar.
However, I see it is not as comfortable as clicking in the web UI.
On the other hand, you can use the integrated scanner tool to scan your code using appropriate hardware or enter the code by hand, which should lead you to the desired result.

jon-daemon, 2024-06-25:
Yes that makes sense. I created a bookmarklet that asks for the ID and loads the part's url. Something like:
```
javascript:(function(){var partid=prompt("Please enter a part id:");if(partid!=null&&partid!=""){var url="partdb.domain/en/part/"+partid;window.open(url,'_blank');}})();
```
I also use alfred on Mac and I haven't thought to add a custom search until now.

closed 2024-06-25 by jon-daemon

RaptorDE, 2024-07-01:
I also think that the search for ID is missing. Entering via the URL is cumbersome, when using the scanner tool I had to write P0 in front of my three-digit part IDs and the Javascript bookmark is more of a workaround than a fix. Especially for inexperienced users it is inconsistent that you can search by MPN and IPN, but not by ID.

d-buchmann, 2024-07-02:
Another approach would be to add the ID column in any table view and sort by that column. I know it's not as efficient as a search would be, but it already helped me in some situations.