Part-DB/Part-DB-server #651: Problem with importing parameters of a Part via CSV

LozanAyip, 2024-07-18:
Im importing parts via CSV, everything is working except the parameters of a Part. It gets imported, but not used. I even exported a existing Part as csv_full and made the same structure but it still doesnt work. Is it even possible and how should I do it? Thanks for the help!

jbtronics, 2024-07-28:
What do you mean by "imported but not used"?

The general problem with CSV is that it is only useful for "flat" data. If you have data with nested information (like multiple parameters for a part), then this is only possible with some tricks (a column for every possible sub entry).

Something like JSON or YAML is a much better fit if you want to import complex data structures.

NikoDelarich, 2024-08-28:
I have the same issue: I can export a component in Full-JSON format and import it back, but the parameters are missing after the import...

Here's what I'm doing:
- Create part with parameters
- Export part ("Full")
- Delete part
- Import part

After the import, the part is created but the parameters are missing. Do I need to change anything in the import file?
I already tried all the different import formats (JSON, CSV, XML, YAML)...

Protoncek, 2025-03-13:
Reviving this...
same problem: i have Part-DB in proxmox, but due to some problems in proxmox installation i've recreated a new LXC and transferred my parts. The problem is that:
- attachments doesn't export from old instance, so i have upload all of them again
- links on all parts, imported from mouser, farnell... are missing:

[image]

[image]

I've tried to export in all possible formats, but result is the same. Any info if i'm doing anything wrong?