Part-DB/Part-DB-server #652: Creation of Categories/Manufacturers/Storage Locations by entering new structure in selection field fails

mindsolve, 2024-07-23:
**Describe the bug**
When adding a new component, one might need to add a new category or new manufacturer to correctly document the part.
The selection fields seem to offer the possibility to create a new category/manufacturer/etc. on-the-fly when entering a text that doesn't match existing objects (prefixed with a plus sign in the appearing drop-down). When selecting the drop-down entry, the value of the field is displayed in bold, as though the value was accepted.

When then trying to save the part, an error message appears (see screenshot below) and the previously selected "new" value is removed and a text appears underneath saying "The selected choice is invalid".

**To Reproduce**
Steps to reproduce the behavior:
1. Click on "New Part"
2. Enter dummy text as name
3. Enter "Dummy category" in Category field, selecting the appearing drop-down item with the same name an prefixed plus (+) symbol
4. Try to save the new item
5. See error

**Expected behavior**
The part and the new category are created.

**Screenshots**
Drop-down entry seemingly suggesting creating a new part:
[image]

Error on part saving:
[image]

**Server Side**
- Part-DB Version: `1.13.1 (master/a8e1171)`
- PHP Version: `PHP 8.1.2-1ubuntu2.18`
- Database Server: `sqlite3`

**Desktop (please complete the following information):**
 - OS: `Windows 10`
 - Browser: `Firefox 128.0`

**Additional context**
- No log lines (in `var/prod.log`) are written when trying to create the part. The HTTP response code is 422.
- I don't know to 100% whether the drop-down item with the plus really is meant as "create new", but if not, it very much intuitively seems that way.

jbtronics, 2024-07-27:
The coloring schema of your Part-DB does not look like 1.13.1.
That is the old color-inversion darkmode that was used before Part-DB 1.5.0 

You should update your Part-DB to the latest version (and ensure your frontend is update-to-date), and I would assume your issue is gone then. In the demo instance, this works fine.

closed 2024-07-27 by jbtronics

mindsolve, 2024-07-27:
Oh... that's strange - sorry for the already solved issue then :/

The main page shows the version as I pasted it above, so I assumed that the update worked.
I even checked from a device that I've never used by PartDB instance from, so it shouldn't be a (browser) cache problem.

Am I missing something in my update flow? I thought I was following the steps from the README exactly:
```bash
# as www-data
cd /var/www/partdb/
git pull
composer install --no-dev -o
php bin/console doctrine:migrations:migrate
php bin/console cache:clear
sudo systemctl restart apache2.service
```

jbtronics, 2024-07-27:
You have to install and build the frontend too (and as far as I know the current documentation mentions this).

That is `yarn install` and `yarn build`.

Afterwards you should clear Part-DB cache with `php bin/console cache:clear`

(you should execute all commands as your Webserver user).

mindsolve, 2024-07-27:
Thank you! That was it!

I think I initially used those steps when debugging the PartKeepr import (some time ago), and there I only needed to update/clear caches for the backend - and seemingly I've blindly reused that until now.