Part-DB/Part-DB-server #657: After the update, the amount column in the parts table will not be shown automatically.

sdouze, 2024-07-29:
**Describe the bug**
After the update, the amount column that I had set up to appear automatically in the tables is lost. I can enable it manually.

My Docker compose yaml configuration.
-TABLE_PARTS_DEFAULT_COLUMNS=name,description,category,storage_location,ipn,manufacturer,footprint,amount

**Screenshots**
[image]
[image]

**Server Side**
- Part-DB Version: Docker 1.13.2
- PHP Version:8.1.29 (SAPI: fpm-fcgi)
- Database Server: sqlite 3.34.1

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 127.0.6533.72

jbtronics, 2024-08-02:
Are you sure that it was not just old customization stored in your browser? You only see the global configuration when you access Part-DB for the first time. Afterwards the columns are shown which were shown last time (and might be customized by the user).

There was no changes related to tables in the last update, so It would be very odd if there would be a change.

sdouze, 2024-08-02:
Hello,

I checked again, and today it works. I do not know what happened. Before the update, the amount column was there. I encountered the problem on the day I updated the partdb.

Thank you for your reply, and I apologize for any confusion.

closed 2024-08-02 by sdouze