Part-DB/Part-DB-server #667: Label generation using IPN

sdouze, 2024-08-10:
**Is your feature request related to a problem? Please describe.**
I get confused when generating labels for part lots because ID on parts and ID for part lots are not the same and I cannot find the Part lot ID.

**Describe the solution you'd like**
I would like to be able generatign labels using IPN instead default ID.

**Describe alternatives you've considered**
Give the option to display part lot ID on the part list table.

jbtronics, 2024-08-23:
> I get confused when generating labels for part lots because ID on parts and ID for part lots

That's because part not necessarily have exactly one part lot. A part can have lots at multiple locations, and you have to choose which lot you want to generate a label for.
Using the IPNs does not fix this problem, as the IPN identifies a part not necessarily a part lot.

> Give the option to display part lot ID on the part list table.

You dont really need to input the lot IDs manually. You can just mark the checkboxes of the parts you want labels for in the part tables, and choose the "Generate labels (for part lots)" option in the menu appearing above the table.

sdouze, 2024-09-04:
> That's because part not necessarily have exactly one part lot. A part can have lots at multiple locations, and you have to choose which lot you want to generate a label for. Using the IPNs does not [...]

Hello, thank you for your response.
Can you share a screenshot about Generate labels option after selecting part because I cannot find this option.
Thank you in advance!