Part-DB/Part-DB-server #677: Search for équivalent parts

anthology, 2024-08-25:
**Is your feature request related to a problem? Please describe.**
When I carry out a repair, I do not necessarily have the exact part but may be an equivalent one.

**Describe the solution you'd like**
It would be perfect to be able to look for an equivalent in the available parts.

some parts sites offer this possibility, is it possible to use it for partdb?

pdo59, 2024-08-28:
I’m actually very interested in a feature like this myself, so your suggestion really resonates with me. Could you please provide examples of sites that offer this functionality? It would help us better understand how they handle equivalent parts and allow us to explore whether such an integration is possible for partdb.

anthology, 2024-08-28:
Partbox have this possibility,
I have tested it quickly, he search for parameter like capacity and voltage for capacitor or hfe, voltage etc for triac
You Can choose what is important and he search

Cons, when he search for the parts, he need a connections from parts suppliers (farnell or other)
For found the parameters of actual components to compare to component in stock

garno, 2024-11-05:
Would love this possibility, even if we have to manually mark parts as equivalent.