Part-DB/Part-DB-server #680: AliExpress as information provider

vergissberlin, 2024-08-27:
**Hi!**

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when i do not find the products to import with the current supported APIs.

**Describe the solution you'd like**
AliExpress  provide an API with product information for affiliate partners.
https://openservice.aliexpress.com/doc/api.htm#/api?cid=21407&path=aliexpress.affiliate.productdetail.get&methodType=GET/POST

**Describe alternatives you've considered**
Typing all my stuff manually.

**Additional context**
Its about the Information provider system.

Example response:
```json
{
  "code": "0",
  "resp_result": {
    "result": {
      "current_record_count": "200",
      "products": [
        {
          "app_sale_price": "300",
          "original_price": "300",
          "product_detail_url": "https://www.aliexpress.com/item/33006951782.html",
          "product_small_image_urls": [],
          "second_level_category_name": "Women\u0027s Clothing",
          "target_sale_price": "320.2",
          "second_level_category_id": "333333",
          "discount": "50%",
          "product_main_image_url": "https://ae01.alicdn.com/kf/HTB1tyl7bELrK1Rjy0Fjq6zYXFXaC/Spring-Autumn-mother-daughter-dress-matching-family-outfits-mother-kids-dress-matching-outfits-flamingo-embroidery-1.jpg",
          "first_level_category_id": "1111",
          "target_sale_price_currency": "USD",
          "target_app_sale_price_currency": "USD",
          "tax_rate": "0.1",
          "original_price_currency": "USD",
          "platform_product_type": "ALL",
          "shop_url": "https://www.aliexpress.com/store/3255036",
          "target_original_price_currency": "USD",
          "product_id": "33006951782",
          "target_original_price": "360.5",
          "product_video_url": "https://ae01.alicdn.com/kf/HTB1tyl7bELrK1Rjy0Fjq6zYXFXaC/Spring-Autumn-mother-daughter-dress-matching-family-outfits-mother-kids-dress-matching-outfits-flamingo-embroidery-1.jpg",
          "first_level_category_name": "dress",
          "promotion_link": "http://s.click.aliexpress.com/e/xxxxx",
          "sku_id": "12000029786932962",
          "evaluate_rate": "89.22%",
          "sale_price": "15.9",
          "product_title": "Spring Autumn mother daughter dress matching family outfits mother kids dress matching outfits flamingo embroidery 1 to 11 yrs",
          "hot_product_commission_rate": "60%",
          "shop_id": "111111",
          "app_sale_price_currency": "USD",
          "sale_price_currency": "USD",
          "lastest_volume": "300",
          "target_app_sale_price": "330.3",
          "commission_rate": "3.5%",
          "promo_code_info": {
            "code_campaigntype": "1",
            "code_availabletime_end": "2020-04-30 23:59:59",
            "code_quantity": "1",
            "code_availabletime_start": "2020-04-01 00:00:00",
            "code_value": "On order over USD 10, get USD 7 off",
            "promo_code": "GMG20207",
            "code_mini_spend": "1",
            "code_promotionurl": "https://s.click.aliexpress.com/e/_xxxxx"
          },
          "relevant_market_commission_rate": "10%"
        }
      ]
    },
    "resp_code": "200",
    "resp_msg": "success"
  },
  "request_id": "0ba2887315178178017221014"
}
```
I konw, it's not a lot but at least we get an title and an image. With https://openservice.aliexpress.com/doc/api.htm#/api?cid=20897&path=aliexpress.solution.sku.attribute.query&methodType=GET/POST it should be possible to get more attributes.

kind regards,
André

riacob, 2025-02-13:
I'm looking for the same solution for managing my inventory, however this issue seems to have gone unnoticed, like, completely.
Hopefully by posting this it will pop up higher in the list...

jbtronics, 2025-02-13:
The problem is that this API is not publicly available (only for seller/vendors on AliExpress), so that is nothing everyone can register for.

That is the same problem with Amazon and eBay.

riacob, 2025-02-13:
Since scraping is impossible due to bot restrictions, the most obvious solution, although not very easy to implement, would be a browser extension that allows you to capture the JSON for a page (i.e. AliX, Amazon, eBay...) and import it into Parts-DB.

jbtronics, 2025-03-31:
I did some experiments with using symfony panther component, which loads the webpage into an headless chrome or firefox instance. As this can execute javascript and looks like a normal browser for Aliexpress, you can load the product page in a way, that allows to scrape infos from that. I have some proof of concept code in the aliexpress-provider branch.

However that way is quite slow, and more difficult to setup than just getting responses and parsing HTML in PHP directly.

Also the number of possible concurrent panther instances is limited (currently), so we probably have to see how useful this would be in practice.

riacob, 2025-04-03:
Did you have any issues with the "human verification" or "slide to verify" popups? Those seemed to be a problem when I was trying to monitor an aliexpress product page with an external monitoring service.

jbtronics, 2025-04-03:
So far not. But i guess that will depend on which IP address is used for scraping and how many requests you are doing.
I would hope, that normal Part-DB use may not trigger the CAPTCHA, because there will be no real workaround possible.