Part-DB/Part-DB-server #681: Unnecessary Rounding of Parameter Values

T0b121, 2024-08-28:
When entering a parameter value, the value gets rounded unnecessarily. I have a 1.5 nF capacitor and want to add an entry for µF for the search. However, after entering the value, it shows 0.002 instead of 0.0015 (see Image 1). When I enter 0.0015 (see Image 2) and press Apply, the cell again shows 0.002 (see Image 3).

Steps to Reproduce:
1. Enter a parameter value of 1.5 nF.
2. Add an entry for µF.
3. Observe that the value is rounded to 0.002 instead of 0.0015 (see Image 1).
4. Manually enter 0.0015 (see Image 2).
5. Press Apply and observe that the value reverts to 0.002 (see Image 3).

Expected Result: The value should remain 0.0015 after entering and applying without any rounding.

Actual Result: The value is rounded to 0.002 after entering and applying.

Attachments:
- Image 1: Shows the value 0.002 after initial entry.
- Image 2: Shows the value 0.0015 before applying.
- Image 3: Shows the value 0.002 after pressing Apply.
- Image 4: Shows the page of the component.

Image 1
[image]

Image 2
[image]

Image 3
[image]

Image 4
[image]

T0b121, 2024-08-28:
Under certain circumstances, the value is not rounded, but written with e-Nutatur (see Image).
[image]

jbtronics, 2024-09-09:
This is now fixed