Part-DB/Part-DB-server #684: Missing child elements after Database restore

AlexD0302, 2024-08-30:
Hello,
is it possible that the described recovery method does not work? 
Let me describe my procedure:

I have created an SQL dump on the old server (this is a Raspberry Pi; mysql -u root -p < backup.sql)

Then I installed the following on the new server:
- env.local
- config/parameters.yaml
- config/banner.md
- uploads/ (entire folder)
- public/media (entire folder)

Finally, I executed the backup with ‘mysql -u root -p < backup.sql’.

Now it sometimes happens that I get the following error, for example when I want to edit or delete components:

Invalid resource provided: ‘/var/www/partdb/translations/messages.zh.xlf’; Errors: [ERROR 1871] Element ‘{urn:oasis:names:tc:xliff:document:2.0}file’: Missing chil
  d element(s). Expected is one of ( {urn:oasis:names:tc:xliff:document:2.0}skeleton, ##other{urn:oasis:names:tc:xliff:document:2.0}*, {urn:oasis:names:tc:xliff:docu
  ment:2.0}notes, {urn:oasis:names:tc:xliff:document:2.0}unit, {urn:oasis:names:tc:xliff:document:2.0}group ). (in /var/www/partdb/ - line 3, column 0)

Is it problematic as I am using Debian 12 on the new server?

jbtronics, 2024-08-30:
This is most likely not an backup/restore problem, but an problem caused by an somehow broken, translation file (it says that the Chinese translation is somewhat broken). That is untouched by the restore process, so it would probably also occur with a fresh database.

You can probably fix it by copying the `translations\messages.zh.xlf` from GitHub into your Part-DB instance. Probably something went wrong with the copying, when you installed Part-DB (but then it's possible that this affects more files)

AlexD0302, 2024-08-31:
Thank you for your answer.
I was able to solve the problem but I don't know how it came about.
Somehow my old server had been updated but obviously not completely....new update to 1.13.3 solved the problem and now everything works fine.

closed 2024-09-09 by jbtronics