Part-DB/Part-DB-server #686: Prevent access to PartsDB for unauthenticated users.

jounathaen, 2024-09-02:
Hi & thanks for all the work put into this project.

I was honestly a little surprised, that you can see all parts, categories etc. in the DB without being an authenticated (logged in) user.  I might have overlooked something, but I couldn't find a setting to change this.
As I'm not super fond of putting a list of our inventory online and accessible for the public, I think it'd be great to prevent the access to the db for unauthenticated users.

Best,
Jonathan

heschdy, 2024-09-02:
Hi,

there is a user called anonymous there you can set the permissions. But keep in mind, when you uncheck the api, that the api will also not work without logging in.

Ralf

jounathaen, 2024-09-02:
Ahh, that is exactly what I'm looking for!
Thanks for the assistance :+1: