Part-DB/Part-DB-server #690: SSO 404 Error page

workino, 2024-09-03:
**Describe the bug**
After the saml authentication the redirected page is wrong landing in a 404 error code.
URL: https://parts.MYDOMAIN.MY/en/saml/acs?RelayState=.......
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to login page
2. Click on SSO Login
3. Authenticate against the sso provider
4. See error

**Expected behavior**
To be authenticate in partsdb with my sso login

**Screenshots**
Screenshot attached before.

**Server Side**
- Part-DB Version: 1.13.3
- PHP Version: partdb docker version
- Database Server [e.g. MySQL 5.7]

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser Chrome
 - Version 128.0.6613.113

workino, 2024-09-04:
Found the problem in the configuration between authentik and part-db.

Part-DB is send as GET and not post so in authentik needs to be used /application/saml/partdb/sso/binding/redirect/ instead of /application/saml/partdb/sso/binding/post/
but the answer is expected as POST. 

SAML_ROLE_MAPPING='{"admin": 1, "editor": 3, "*": 2}' is not optional so need to be set according to the group in authentik.

workino, 2024-09-04:
It's working perfectly with authentik

closed 2024-09-04 by workino

heschdy, 2024-09-04:
Hi, I'm really struggling to set authentik up. Can you share how you have configured it?

workino, 2024-09-04:
Hello Heschdy,
In authentik the SAML protocol need to be configured like:
[image]

and the advanced part:
[image]

I created a new certificate specifically for pertdb installation. 
NameID Property Mapping needs to be set as your username or by default will be the ID
Rest is by default.

In the .env file (I have it installed in docke) the parameters are: 
SAMLP_SP_PRIVATE_KEY=MII...........
SAML_SP_ENTITY_ID=http://PARTSDB.YOURDOMAIN.TLD/sp
SAML_IDP_ENTITY_ID=https://sso.YOURDOMAIN.TLD
SAML_IDP_SINGLE_SIGN_ON_SERVICE=https://sso.YOURDOMAIN.TLD/application/saml/partdb/sso/binding/redirect/
SAML_IDP_SINGLE_LOGOUT_SERVICE=https://sso.YOURDOMAIN.TLD/application/saml/partdb/slo/binding/redirect/
SAML_IDP_X509_CERT=MII............
DEFAULT_URI=https://PARTSDB.YOURDOMAIN.TLD/  (pay attention to have the / at the end )
SAML_ROLE_MAPPING='{"admin": 1, "editor": 3, "*": 2}'
SAML_ENABLED=true

And it will work. 
Regards
Workino :)

heschdy, 2024-09-04:
Thanks a lot. 
Authentifikation is now working. The only problem is that the role mapping is not working. I have added the groups in the property mapping as in your screenshot. If I test it inside authentik provider I'm also seeing that the user is part of the admin group. But the user is always in the readonly group.

workino, 2024-09-04:
As you can see my user is part of these 2 groups
[image]
Just pay attention is written admin and not adminS. The rule is referring to the one without S