Part-DB/Part-DB-server #694: Add distributor from an information provider

mariosgit, 2024-09-09:
Creating parts from info providers is a cool feature...

**~The problem~ One more use for this**

It would be nice to add a distributor to an existing part by searching the part @ a provider..

So I have a part created from mouser. Now I would like to add JLC or whatever as a second distributor. In that case a search @ JLC would speed up the process a lot, copying the supplier part number and the link and prices, images...

**Describe the solution you'd like**

Next to PurchaseInfo / AddDistributor a button to search the same part  @ other info providers.

Cheers.

jbtronics, 2024-09-09:
You can already do this. In the tools tab for a part detail, there is a button "Update part from info providers", which then redirects you to the provider search page. There you can select a provider part and this info will be merged into the existing part.
Meaning that empty fields will be filled and things like distributor infos are added.

mariosgit, 2024-09-09:
Ah ok, didn't see that. Thanks.

closed 2024-09-09 by mariosgit