Part-DB/Part-DB-server #708: Save things you changed on the ui (e.g. the listed attributes of the parts in the part list) after restarting docker

CodingFactoryT, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
If I restart docker, all settings that I set beforehand (like including the edit button in the parts list) are lost.

**Describe the solution you'd like**
Save these settings/filters.

**Describe alternatives you've considered**
-
**Additional context**
-

jbtronics, 2024-09-14:
These things should already be saved in the localstorage of your browser.

You must not clear the browser data for the Part-DB page, or these settings will be lost.

CodingFactoryT, 2024-09-14:
Hmm, strange. I didn´t clear the browser data manually, but each time I restart my computer and restart docker, the settings get lost.