Part-DB/Part-DB-server #713: How to fork this project and add my own features

CodingFactoryT, 2024-09-19:
Hey,
I'd like to gork this project in order to create some special features I need, but I don't know how to set the project up for development after downloaden the code. Could you help me a bit with that? What programs do i need, how do I host the server for development and how do I dockerize it when I'm finished?
I'd be happy for a response, Thanks!

jbtronics, 2024-09-23:
You need the normal requirements for Part-DB (PHP, composer, maybe a database, nodejs & yarn). In principle you can use a full webserver for development, but I find the [symfony-cli server](https://symfony.com/download) easier to use, as it needs no configuration.
As an IDE I use PHPstorm with the symfony plugin, but you can use whatever you find good to use as an code editor.

You can build an docker image using one of the Dockerfiles in the project root.

CodingFactoryT, 2024-09-25:
When running yarn build, I encounter this error:

Failed to compile with 2 errors                                                                         11:14:17

Module build failed: Module not found:
"./assets/translator.js" contains a reference to the file "../var/translations/configuration".
This file can not be found, please check it for typos or update it if the file got moved.

"./assets/translator.js" contains a reference to the file "../var/translations".
This file can not be found, please check it for typos or update it if the file got moved.

webpack compiled with 2 errors and 9 warnings

Indeed, the var folder doesn´t contain a translations folder nor a configuration file for translations.

Can you help me with that, @jbtronics ? Thanks!

CodingFactoryT, 2024-09-25:
I managed to make it run, thank you :-)

closed 2024-09-27 by CodingFactoryT