Part-DB/Part-DB-server #717: Users cant configure 2FA

Jamsterz, 2024-09-27:
Permissions issue only allows members of the Admins group to configure 2FA.

Other users should be able to configure 2FA

Jamsterz, 2024-10-01:
> Every (non SAML) user should be able to add 2FA (like he should be able to change his password). What does happen when a user tries to do this?

It lets you put in the code and does nothing after. If you put in the wrong code it tells you it’s wrong but putting in the correct code does nothing. Changing the users groups from Users to Admins the correct code adds the Authenticator successfully

jbtronics, 2024-10-13:
This was caused by an constraint that should prevent users to lockout themselves accidentally, which was triggered here for the wrong reason. This is fixed with the latest commits.

closed 2024-10-13 by jbtronics