Part-DB/Part-DB-server #718: Unable to import Parameters on parts even when using JSON

rumblesan, 2024-09-27:
Hi,

I'm trying to use the import functionality to populate a new install of Part DB from an existing spreadsheet I have. I've written a python script to format the information into JSON so I can attempt to import a bunch of the information as parameters on parts, but it doesn't seem to be working. New parts are created, but there aren't any parameters on them.

I found issue #363 where somebody seemed to have the same problem and it was apparently resolved but that doesn't seem to be the case?

I'm running Part DB locally using docker, and I've tried both the latest image from docker hub and one that I built myself from the latest commit on the github repo but no difference.

The format for the part JSON seems to be correct based on what I understand, but maybe that's not the case?

```
[
  {
    "name": "5pF ceramic C0G",
    "category": "Capacitor->Through Hole->Ceramic",
    "footprint": "C_Rect_L7.0mm_W3.5mm_P5.00mm",
    "manufacturer": "TDK",
    "supplier": "mouser",
    "parameters": [
      {
        "name": "Dielectric",
        "value_text": "ceramic C0G",
        "formatted": "ceramic C0G",
        "_type": "Part"
      }
    ]
  }
]
```

Part DB seems like a great piece of software and looks to hit exactly the right levels of functionality vs complexity for my usage, so would really like to get using it properly.
Appreciate any help you can give.

cheers

jbtronics, 2024-10-13:
There was indeed a bug with parameter import on parts. The serialization group, that says which properties should be importable was missing for the parts parameters. 

That is fixed with the latest commits.

closed 2024-10-13 by jbtronics

rusucc, 2024-11-26:
Hello! Is your capacitor example currently functional ? I am receiving "invalid category" errors.