Part-DB/Part-DB-server #721: Label QR Code only contains IP address

Bubu5485, 2024-10-03:
Dear all,

I have a problem when generating a Label in the database.

No matter what I choose and what I put in for data I only get the IP address after reading the QR Code.

I tried in Twig mode and Placeholders mode no change.

Is there a setting which I need to turn on or the system works like that?

Thank you very much for the help.

anthology, 2024-10-04:
Hi
You generate QR code on the menu tools/label profiles or on a component?
When i test a label profiles, i have juste the IP or my domaine but the complete URL on component label generation

Bubu5485, 2024-10-04:
Hi,

I tried on both and I get the same results. Is there a work around to get something else on the label?

anthology, 2024-10-04:
Yes, you Can add component id, storage and many more

Bubu5485, 2024-10-04:
I only get in the result is the server IP when I read the QR Code. :(

jbtronics, 2024-10-04:
Do you access Part-DB via an IP address? Or via a reverse proxy?
If so have you checked, that the reverse proxy is correctly configured (especially on the part-db side with `TRUSTED_PROXIES`?)

The QR codes should contain a URL, where the host part is based on the one which you use to access the Part-DB website. When the QR code URL is odd, then there is most likely an issue with the webserver or reverse proxy configuration, which makes it impossible to resolve the correct URL.

anthology, 2024-10-04:
Can you post a qrcode for test?Your qrcode reader is good ?

Bubu5485, 2024-10-05:
> Do you access Part-DB via an IP address? Or via a reverse proxy?
> If so have you checked, that the reverse proxy is correctly configured (especially on the part-db side with `TRUSTED_PROXIES`?)
>
> The [...]

I access Partdb via IP.

Bubu5485, 2024-10-05:
I will post it on Monday. Im using my phones camera to read the qrcode.

Bubu5485, 2024-10-07:
> Can you post a qrcode for test? Your qrcode reader is good ?
[qrcode.pdf](https://github.com/user-attachments/files/17274087/qrcode.pdf)

There you go.

Bubu5485, 2024-10-08:
> Can you post a qrcode for test? Your qrcode reader is good ?

Hi,

Just tested the QR Code and it takes me to the part so its working perfectly.

Thanks for the help.