Part-DB/Part-DB-server #722: Troubles with running initial migrations against PostgreSQL deployed via Debian packages

plaes, 2024-10-04:
When attempting to run the initial migrations (1.1. against PostgreSQL hosted on Debian, I get a "Invalid platform version "" specified. The platform version has to be specified in the format: "<major_version>.<minor_version>.<patch_version>" error.

It seems that dbal expects reported postgresql version to be in certain format, though at least on Debian hosts the versions are following:

This one is from PostgreSQL instance deployed in kubernetes via cnpg operator:
```
partdb=> SELECT version();
                                                           version                                                           
-----------------------------------------------------------------------------------------------------------------------------
 PostgreSQL 16.4 (Debian 16.4-1.pgdg110+2) on x86_64-pc-linux-gnu, compiled by gcc (Debian 10.2.1-6) 10.2.1 20210110, 64-bit
(1 row)

partdb=> SHOW server_version;
         server_version         
--------------------------------
 16.4 (Debian 16.4-1.pgdg110+2)
```

And another (just plain Debian installed on bare-metal server):
```
postgres=# show server_version;
        server_version        
------------------------------
 15.8 (Debian 15.8-0+deb12u1)
(1 row)

postgres=# select version();
                                                      version                                                      
-------------------------------------------------------------------------------------------------------------------
 PostgreSQL 15.8 (Debian 15.8-0+deb12u1) on x86_64-pc-linux-gnu, compiled by gcc (Debian 12.2.0-14) 12.2.0, 64-bit
(1 row)
```

**To Reproduce**
Steps to (probably) reproduce the behavior:
1. Set up Debian with PostgreSQL installed
2. Attempt to run initial migrations against this database

**Expected behavior**
Ideally, dbal shouldn't parse the string version, instead it should use `SELECT current_setting('server_version_num');`

I was able to work around this issue by temporarily editing the `/var/www/html/vendor/doctrine/dbal/src/Driver/AbstractPostgreSQLDriver.php` and "fixing" the version manually there...

jbtronics, 2024-10-04:
What happens if you specify the server version of the database manually in the `DATABASE_URL` env? (without these debian part?):
```
DATABASE_URL="postgresql://db_user:db_password@127.0.0.1:5432/db_name?serverVersion=12.19&charset=utf8"
```
The exact version given there should not really matter for normal usage.

> Ideally, dbal shouldn't parse the string version, instead it should use SELECT current_setting('server_version_num');

That is nothing Part-DB can change, you should maybe open an issue at DBAL for this.

plaes, 2024-10-05:
> What happens if you specify the server version of the database manually in the `DATABASE_URL` env? (without these debian part?):
>
> ``` [...]

Yeah, I tried searching similar issue in DBAL issue tracker first, but nothing popped up, so decided to first report it here in case it's happening due to the way application is setting things up.

Specifying `serverVersion` probably works, though as I'm using [semi-automated kubernetes setup](https://gist.github.com/plaes/4ba335a30caafc12b1af6a297ab86cc5), I would be interested to get this fixed without workarounds.

plaes, 2024-10-08:
Apparently it's a "feature" of Symfony Cache - https://github.com/symfony/symfony/discussions/58491

closed 2024-10-08 by plaes