Part-DB/Part-DB-server #730: Frustrating Error 500 without much details

sochartetu, 2024-10-15:
"GET /en/part/30/info HTTP/2.0" 500 3467 "https://192.168.10.10:12443/en/category/33/parts" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:131.0) Gecko/20100101 Firefox/131.0"

Get Error 500 when I try to get info about parts.
Not a lot of details in docker logs.

sochartetu, 2024-10-15:
So, after investing 5h to debug that, I've found out the culprit
is mb_strimwidth.
Despite mbstring being installed and enable in php 8.2 fpm, I get an error when going to Dev mode.
When commented, no more error.
WTH ?

[image]

jbtronics, 2024-10-16:
What is the exact error message for this function to fail?

Do you have some weirdly encoded text in your database? I see not much reasons why this specific function should fail. Especially as these function has no (documented) failure behavior, which could cause an exception.

sochartetu, 2024-10-17:
Fatal error: Call to undefined function mb_strimwidth() in ...

mbstring module is enable in PHP FPM and PHP CLI

No special char.

CyberTeckyLLC, 2024-10-19:
I'm not sure if this is related, but I've included another 500 internal server error while running docker-compose.yml.  The following 'docker-error.md' contains the docker-compose logs -f info.

[docker-error.md](https://github.com/user-attachments/files/17442772/docker-error.md)

[image]

jbtronics, 2024-11-03:
> I'm not sure if this is related, but I've included another 500 internal server error while running docker-compose.yml. The following 'docker-error.md' contains the docker-compose logs -f info.
> 
> [docker-error.md](https://github.com/user-attachments/files/17442772/docker-error.md)
> 
> ![Server 500 Internal Error](https://private-user-images.githubusercontent.com/99003155/378052225-505e2036-5e80-4f2c-ab56-67de30193fe6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NzAyMzAsIm5iZiI6MTczMDY2OTkzMCwicGF0aCI6Ii85OTAwMzE1NS8zNzgwNTIyMjUtNTA1ZTIwMzYtNWU4MC00ZjJjLWFiNTYtNjdkZTMwMTkzZmU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAzVDIxMzg1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViNWVlNWJhZmM5ODNiMjk2ZWI5MmUyZGM0YzJjZTI1MjFjMjNkZGVjYmZlNzJmZjA0NmU4YWQ5YmE0Y2U2ZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2Jj7Xk4SVDPBeeyDNeZr-y4zORbiYI6VCiXjhkbAOIU)

This error log just tells you that Part-DB cannot reach the database server and cannnot resolve its name (probably due to some misconfiguration somewhere). Thats unrelated to this issue here

closed 2025-01-05 by jbtronics