Part-DB/Part-DB-server #734: Internal error 500 when sorting on Storage locations column

crazyelectron-io, 2024-10-16:
**Describe the bug**
When clicking on the 'Storage locations' column header to sort on location, an internal error 500 is shown: `Error calling [/en/store_location/222/parts](https://parts.MYDOMAIN.TLD/en/store_location/222/parts).` When opening the details it also shows _Infos for Admin: Error while executing database query._
Other columns I checked work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a list of parts
2. Click on 'Storage locations' table header
3. See error

**Expected behavior**
The storage location column sorted alphabetical.

**Server Side**
- Part-DB Version: 1.14.0
- Official Docker image on Debian Linux
- Database Server Postgres 16.4

**Desktop (please complete the following information):**
 - OS: MacOS 15
 - Browser Chrome
 - Version 129.0.6668.100

jbtronics, 2024-10-16:
What does the server log say?

crazyelectron-io, 2024-10-17:
Sorry, I thought I added a log extract. This is logged:
```
10.42.3.129 - - [17/Oct/2024:05:08:10 +0000] "POST /en/store_location/227/parts HTTP/1.1" 200 33378 "https://parts.EXAMPLE.TLD/en/store_location/227/parts" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/129.0.0.0 Safari/537.36"
{"message":"Matched route \"part_list_store_location\".","context":{"route":"part_list_store_location","route_parameters":{"_route":"part_list_store_location","_controller":"App\\Controller\\PartListsController::showStorelocation","_locale":"en","id":"227"},"request_uri":"https://parts.EXAMPLE.TLD/en/store_location/227/parts","method":"POST"},"level":200,"level_name":"INFO","channel":"request","datetime":"2024-10-17T05:08:13.191754+00:00","extra":{"token":null,"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Read existing security token from the session.","context":{"key":"_security_main","token_class":"Symfony\\Component\\Security\\Core\\Authentication\\Token\\UsernamePasswordToken"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.192703+00:00","extra":{"token":null,"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"User was reloaded from a user provider.","context":{"provider":"Symfony\\Bridge\\Doctrine\\Security\\User\\EntityUserProvider","username":"admin"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204900+00:00","extra":{"token":null,"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Checking for authenticator support.","context":{"firewall_name":"main","authenticators":5},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204919+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"App\\Security\\ApiTokenAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204929+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"App\\Security\\ApiTokenAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204937+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204940+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204947+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204949+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2024-10-17T05:08:13.204953+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer [... cut ...] :"/en/store_location/227/parts","ip":"10.42.2.0","http_method":"POST","server":"parts.EXAMPLE.TLD","referrer":"https://parts.EXAMPLE.TLD/en/store_location/227/parts"}}
10.42.3.129 - - [17/Oct/2024:05:08:13 +0000] "POST /en/store_location/227/parts HTTP/1.1" 500 4576 "https://parts.EXAMPLE.TLD/en/store_location/227/parts" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/129.0.0.0 Safari/537.36"
- -  17/Oct/2024:05:08:13 +0000 "POST /index.php" 500
```

jbtronics, 2024-11-04:
Should be fixed now

closed 2024-11-04 by jbtronics