Part-DB/Part-DB-server #742: OEMSecrets Information Provider could not be activated

mariobaier26, 2024-10-24:
**Describe the bug**
I've set a API-Key from OEMSecrets, but it could not be activated. It seems, as the PHP-Part of that Provider never reacted to the environment variables. I use the official Docker Container jbtronics/part-db1 on a raspberry pi

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the docker-compose.yml file and put the PROVIDER_OEMSECRETS_KEY variable in, with a appropiate key and run `docker compose up -d`
2. Go on the instance-website, login and go to **tools ->  Create parts from info provider** on the left navigation
3. Click on the Link **Show all available info providers** to see that OEMSecrets is still disabled

**Expected behavior**
It should be enabled, after put the environment variable in and recreate the container with docker

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.14.2
- PHP Version: 8.2.24 (SAPI: fpm-fcgi)
- Database Server: sqlite 3.40.1
- Device: Raspberry Pi 3B+ wtih Debian 11, aarch64

**Desktop (please complete the following information):**
 - OS: Debian 11
 - Browser Chromium
 - Version 129.0.6668.89

**Additional context**
I also enabled TME and LCSC, which work flawlessly, but OEMSecrets has more articles and information in it's database. LCSC gets disabled, if that works. If i missed some Information, please tell me what you need.

jbtronics, 2024-11-03:
The env variables were not passed correctly to Part-DB. This should be fixed with the latest commits.

mariobaier26, 2024-11-05:
A huge thank you for your Work. The Search with the OEMsecrets API works now.

sochartgit, 2024-11-14:
Despite using last version v1.14.4, and ENV variable for OEMSECRETS properly filled and passed to the container, I do have the exact same problem as @mariobaier26.

mariobaier26, 2024-11-18:
@sochartgit i updated my docker container to v1.14.4 and checked again. it still works with OEMSECRETS on my side without any issue.

closed 2025-01-05 by jbtronics