Part-DB/Part-DB-server #748: Unable to get, or change, admin password with Docker installation using MySQL 8

BobCumberbatch, 2024-10-27:
**Describe the bug**
Installed using instructions here: https://docs.part-db.de/installation/installation_docker.html

After executing command: docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate

I receive the following: 
$ docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate

WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > 

 [OK] Already at the latest version ("DoctrineMigrations\Version20240728145604")

Note the absence of the admin password.
---
If I try to change the admin password using:

$ docker exec --user=www-data partdb php bin/console partdb:users:set-password admin

I get this error message:

! [NOTE] User found!                                                           

 You are going to change the password of  (@admin) with ID 2. Proceed? (yes/no) [yes]:
 > {"message":"User Deprecated: Support for MySQL < 8 is deprecated and will be removed in DBAL 5 (AbstractMySQLDriver.php:70 called by AbstractDriverMiddleware.php:32, https://github.com/doctrine/dbal/pull/6343, package doctrine/dbal)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Support for MySQL < 8 is deprecated and will be removed in DBAL 5 (AbstractMySQLDriver.php:70 called by AbstractDriverMiddleware.php:32, https://github.com/doctrine/dbal/pull/6343, package doctrine/dbal)","code":0,"file":"/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:210","trace":["/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:108","/var/www/html/vendor/doctrine/dbal/src/Driver/AbstractMySQLDriver.php:70","/var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php:32","/var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php:32","/var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php:32","/var/www/html/vendor/doctrine/doctrine-bundle/src/ConnectionFactory.php:122","/var/www/html/var/cache/docker/ContainerQDJaPXQ/App_KernelDockerContainer.php:269","/var/www/html/var/cache/docker/ContainerQDJaPXQ/App_KernelDockerContainer.php:340","/var/www/html/var/cache/docker/ContainerQDJaPXQ/App_KernelDockerContainer.php:301","/var/www/html/vendor/symfony/var-exporter/Internal/LazyObjectState.php:100","/var/www/html/vendor/symfony/var-exporter/LazyGhostTrait.php:178","/var/www/html/vendor/doctrine/orm/src/EntityManager.php:495","/var/www/html/src/Command/User/SetPasswordCommand.php:58","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1096","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:126","/var/www/html/vendor/symfony/console/Application.php:324","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:175","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:49","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:18"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2024-10-27T18:21:49.809120+00:00","extra":{"token":null,"command":{"name":"partdb:users:set-password","arguments":{"command":"partdb:users:set-password","user":"admin"}}}}
{"message":"Error thrown while running command \"partdb:users:set-password admin\". Message: \"Symfony\\Component\\PasswordHasher\\Hasher\\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97\"","context":{"exception":{"class":"TypeError","message":"Symfony\\Component\\PasswordHasher\\Hasher\\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97","code":0,"file":"/var/www/html/vendor/symfony/password-hasher/Hasher/UserPasswordHasher.php:33","trace":["/var/www/html/src/Command/User/SetPasswordCommand.php:97","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1096","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:126","/var/www/html/vendor/symfony/console/Application.php:324","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:175","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:49","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:18"]},"command":"partdb:users:set-password admin","message":"Symfony\\Component\\PasswordHasher\\Hasher\\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97"},"level":500,"level_name":"CRITICAL","channel":"console","datetime":"2024-10-27T18:21:49.879150+00:00","extra":{"token":null,"command":{"name":"partdb:users:set-password","arguments":{"command":"partdb:users:set-password","user":"admin"}}}}
18:21:49 CRITICAL  [console] Error thrown while running command "partdb:users:set-password admin". Message: "Symfony\Component\PasswordHasher\Hasher\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97" ["exception" => TypeError { …},"command" => "partdb:users:set-password admin","message" => "Symfony\Component\PasswordHasher\Hasher\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97"] ["token" => null,"command" => ["name" => "partdb:users:set-password","arguments" => ["command" => "partdb:users:set-password","user" => "admin"]]]
{"message":"Command \"partdb:users:set-password admin\" exited with code \"1\"","context":{"command":"partdb:users:set-password admin","code":1},"level":100,"level_name":"DEBUG","channel":"console","datetime":"2024-10-27T18:21:49.887883+00:00","extra":{"token":null,"command":{"name":"partdb:users:set-password","arguments":{"command":"partdb:users:set-password","user":"admin"}}}}
{"message":"Uncaught Error: Symfony\\Component\\PasswordHasher\\Hasher\\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97","context":{"exception":{"class":"TypeError","message":"Symfony\\Component\\PasswordHasher\\Hasher\\UserPasswordHasher::hashPassword(): Argument #2 ($plainPassword) must be of type string, null given, called in /var/www/html/src/Command/User/SetPasswordCommand.php on line 97","code":0,"file":"/var/www/html/vendor/symfony/password-hasher/Hasher/UserPasswordHasher.php:33","trace":["/var/www/html/src/Command/User/SetPasswordCommand.php:97","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1096","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:126","/var/www/html/vendor/symfony/console/Application.php:324","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:175","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:49","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:18"]}},"level":500,"level_name":"CRITICAL","channel":"php","datetime":"2024-10-27T18:21:49.888175+00:00","extra":{"token":null,"command":{"name":"partdb:users:set-password","arguments":{"command":"partdb:users:set-password","user":"admin"}}}}

In UserPasswordHasher.php line 33:
                                                                               
  Symfony\Component\PasswordHasher\Hasher\UserPasswordHasher::hashPassword():  
   Argument #2 ($plainPassword) must be of type string, null given, called in  
   /var/www/html/src/Command/User/SetPasswordCommand.php on line 97            
                                                                               

partdb:users:set-password <user>

**Expected behavior**
admin password must be shown during database initialisation and must be able to be changed

**Server Side**
- Part-DB Version: latest when installed using Docker
- PHP Version:latest when installed using Docker
- Database Server latest when installed using Docker

**Desktop (please complete the following information): Raspberry Pi** 
 - OS:Pi OS 64bit
 - Browser Firefox
 - Version latest

**Additional context**
Seems fine if I use SQLite. I have tried clearing cache and several attempts to reinstall without success using MySQL.

jbtronics, 2024-11-03:
> I receive the following: $ docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate
>
> WARNING! You are about to execute a migration in database "partdb" that could result in [...]

This means the initial database migration is already done. Probably you have run the command before and missed the password.

The problems with the password change command, are most likely caused by the missing `-it` (to allow for an interactive TTY) option of `docker exec`. Without it the commands cannot receive user input and you cannot enter a password. That a decision on how docker exec works, Part-DB cannot change that..

With `docker exec -it --user=www-data partdb php bin/console partdb:users:set-password admin` it will probably work.

BobCumberbatch, 2024-11-04:
Hi Jan @jbtronics,
Thank you so much for replying. I was able to set a new password using `docker exec -it --user=www-data partdb php bin/console partdb:users:set-password admin` and I have successfully logged in.

closed 2024-11-04 by BobCumberbatch