Part-DB/Part-DB-server #749: Internal error 500 because of Malformed UTF-8 character

mkne, 2024-10-29:
**Describe the bug**
When trying to edit Suppliers or Footprints I get an error 500. Docker Logs have the following entry with a potential cause of a Malformed UTF-8 character.

{"message":"Uncaught PHP Exception Twig\\Error\\RuntimeError: \"An exception has been thrown during the rendering of a template (\"Malformed UTF-8 characters, possibly incorrectly encoded\").\" at base_admin.html.twig line 31","context":{"exception":{"class":"Twig\\Error\\RuntimeError","message":"An exception has been thrown during the rendering of a template (\"Malformed UTF-8 characters, possibly incorrectly encoded\").","code":0,"file":"/var/www/html/templates/admin/base_admin.html.twig:31","trace":["/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:168","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:80","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/87/8773eeacc1eac17d3fb93d41b570a668.php:297","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/87/8773eeacc1eac17d3fb93d41b570a668.php:138","/var/www/html/vendor/twig
/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:56","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/71/716628020905261bb24296cad242e7ca.php:66","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/84/84f6d5ce79e3f166a25c186180100862.php:53","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/vendor/twig/twig/src/Template.php:349","/var/www/html/vendor/twig/twig/src/Template.php:364","/var/www/html/vendor/twig/twig/src/TemplateWrapper.php:35","/var/www/html/vendor/twig/twig/src/Environment.php:306","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:448","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:453","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:253","/var/www/html/src/Controller/AdminPages/BaseAdminController.php:380","/var/www/html/src/Controller/AdminPa
ges/FootprintController.php:70","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:181","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"],"previous":{"class":"JsonException","message":"Malformed UTF-8 characters, possibly incorrectly encoded","code":5,"file":"/var/www/html/src/Twig/EntityExtension.php:94","trace":["/var/www/html/src/Twig/EntityExtension.php:94","/var/www/html/src/Twig/EntityExtension.php:74","/var/www/html/var/cache/docker/twig/d6/d6bc2befe2e15064534f2f6fb2fe5097.php:198","/var/www/html/vendor/twig/twig/src/Extension/CoreExtension.php:1919","/var/www/html/var/cache/docker/twig/d6/d6bc2befe2e15064534f2f6fb2fe5097.php:193","/var/www/html/vendor/twig/twig/src/Extension/CoreExtension.php:1293","/var/www/html/var/cache/docker/twig/71/716628020905261bb24296cad242e7c
a.php:133","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:168","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:80","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/87/8773eeacc1eac17d3fb93d41b570a668.php:297","/var/www/html/vendor/twig/twig/src/Template.php:437","/var/www/html/var/cache/docker/twig/87/8773eeacc1eac17d3fb93d41b570a668.php:138","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/b1/b1afd000c3870fe5195f52ecbae82be9.php:56","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/71/716628020905261bb24296cad242e7ca.php:66","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/html/var/cache/docker/twig/84/84f6d5ce79e3f166a25c186180100862.php:53","/var/www/html/vendor/twig/twig/src/Template.php:393","/var/www/
html/vendor/twig/twig/src/Template.php:349","/var/www/html/vendor/twig/twig/src/Template.php:364","/var/www/html/vendor/twig/twig/src/TemplateWrapper.php:35","/var/www/html/vendor/twig/twig/src/Environment.php:306","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:448","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:453","/var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php:253","/var/www/html/src/Controller/AdminPages/BaseAdminController.php:380","/var/www/html/src/Controller/AdminPages/FootprintController.php:70","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:181","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"]}}},"level":500,"level_name":"CRITICAL","channel":"request","datetime"
:"2024-10-29T14:44:12.051693+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/de/footprint/new","ip":"172.16.1.75","http_method":"GET","server":"xxx.local","referrer":"http://xxx.local:8080/de/tools/builtin_footprints"}}

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Tools-->Edit-->Supplier
or
1. Click on Tools-->Edit-->Footprint

**Expected behavior**
Being able to edit the Suppliers or the Footprints

**Server Side**
- Part-DB Version: Current Docker Image
- PHP Version: Current Docker Image
- Database Server SQLite

**Desktop:**
 - OS: Windows 11
 - Browser Chrome
 - Version 130.0.6723.70

**Additional context**
I have imported some parts from as csv list using the csv-Example. There might have been non UTF-8 characters in the list. As some of the suppliers name ("Bürklin") have a strange character instead of the "ü".

mkne, 2024-10-31:
[image]
I could fix the issue directly in the database by replacing the supplier name cell with a UTF-8-Text value instead of the BLOB value.

jbtronics, 2024-11-03:
I assume this happened because you imported a file in ISO-8859-1 encoding instead of UTF-8. I do not know how this was possible, as in my experiments I was not able to do this...

As you have fixed it yourself now, I have at least added some encoding conversion logic to importing, that should hopefully avoid such issues in the future.

closed 2024-11-03 by jbtronics