Part-DB/Part-DB-server #750: API search by "tags" is not returning all components

balbet, 2024-10-30:
**Describe the bug**

We are using the "Tags" field to store our internal component part number reference. Several components might have the same reference, because they could be equivalent (for example, 2 resistors with the same value, package and tolerance, but with different MPN). It's nice that we can add several tags, the problem comes when we try to retrieve them using the API:

https://our_database/api/parts?tags=Z05959

This only returns the component that matches exactly with the tag Z05959. I can see it in this format the JSON:
"tags":"Z05959"
But the component that has 2 tags was stored as comma separated. I can see it in this format the JSON:
"tags":"Z06108,Z05959"

So, my API request doesn't return the last component. The only way to get it is sending this request:
https://our_database/api/parts?tags=Z06108,Z05959

But of course, this is not an option because we don't know the other reference.

**Expected behavior**

The expected behavior would be that we get all the components where the tag matches in each comma separated string, and not the entire string. This works fine when the search is done using the web:

https://our_database/en/parts/search?keyword=Z05959&name=1&category=1&description=1&mpn=1&tags=1&storelocation=1&comment=1&ipn=1&ordernr=1

That's why we thing the API behaviour is a bug.

Thanks!

jbtronics, 2024-11-04:
In general, these text filters work by trying to match the text exactly. You need to add LIKE wildcards (% for multiple characters, and _ for one character) to find also substrings.

The way the tags are stored internally makes it indeed a bit difficult to filter this. I have added an special filter for the tags tag, which does all the required queries in background. So now you can just supply the tag and you will receive all parts which contains have this tag.

closed 2024-11-04 by jbtronics

balbet, 2024-11-11:
Perfect, thanks Jan! So, when you say "I have added an special filter...", is this fix already available? Or should we wait until the next version release?

Thanks!

jbtronics, 2024-11-11:
It's part of the already released Part-DB 1.14.3

balbet, 2024-11-11:
Great, thanks for your fast support! :)

balbet, 2024-11-18:
Sorry about writing to a closed ticket, but I talked to IT and they have no clue about how to update it. Do you have some guide? I saw a migration procedure, but I think it only applies to older versions (we currently have the 1.11.3). Thank you in advance!

jbtronics, 2024-11-18:
See here: 

https://docs.part-db.de/installation/installation_guide-debian.html#update-part-db

or here:

https://docs.part-db.de/installation/installation_docker.html#direct-use-of-docker-image

Depending whether you use docker or a direct installation