Part-DB/Part-DB-server #778: KiCAD BOM importer fails with project where part has multiple footprints

plaes, 2024-11-22:
**Describe the bug**
KiCAD's BOM import feature seems to break down whenever a board is using components for which there are multiple footprints used on the board as I can have only unique components in the PartDB BOM.

For example, if I have a board which has multiple 4.7uF capacitors, but I use at least two different footprints for placement:
```csv
4;"C7,C6,C12";"C_0805_2012Metric_Pad1.15x1.40mm_HandSolder";3;"4.7uF";;;
19;"C16,C15,C10,C21";"C_0805_2012Metric_Pad1.18x1.45mm_HandSolder";4;"4.7uF";;;
```

**To Reproduce**
Steps to reproduce the behavior:
1. Fetch the BlueSCSI 50pin desktop project from https://github.com/BlueSCSI/BlueSCSI-v2/tree/main/hardware/Desktop_50_Pin and generate BOM in the PCB editor
2. Import BOM...
3. Assign that 4.7uF component to lines 4 and 19
4. Attempt to save...

**Expected behavior**
Not really sure what would be good solution here...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: [v1.14.4](https://github.com/Part-DB/Part-DB-server/releases/tag/v1.14.4)
- PHP Version: (the one that comes with docker image)
- Database Server: PostgreSQL

jbtronics, 2024-12-01:
I would say that the actual KiCAD footprints are not that relevant for Part-DB. As long as the same part is used for the different footprint variants, it does not really matter for Part-DB project system and you can just unify both BOM entries into a single one.

This is something you will need to do by hand however, as its not really possible to detect what parts you really intend with the value and the case that different footprints mean different components is the more common case.

plaes, 2024-12-02:
> I would say that the actual KiCAD footprints are not that relevant for Part-DB. As long as the same part is used for the different footprint variants, it does not really matter for Part-DB project [...]

How hard would it be to support single component multiple times in BOM (or some kind of "merged" rows feature)?

If I take a look at current [BOM output - Desktop_50_Pin_TopConn.csv](https://github.com/user-attachments/files/17975812/Desktop_50_Pin_TopConn.csv) of BlueSCSI-v2, then  this one has at least 5 of those "annoying" components:
1. `0.1uF` -> 3 rows (`C_0805_2012Metric`, `C_0805_2012Metric_Pad1.18x1.45mm_HandSolder` and `C_0805_2012Metric_Pad1.15x1.40mm_HandSolder`
2. `1k` -> 2 rows (`R_0402_1005Metric` and `R_0402_1005Metric_Pad0.72x0.64mm_HandSolder`)
3. `20uF` -> 2 rows (`C_1206_3216Metric` and `C_1206_3216Metric_Pad1.42x1.75mm_HandSolder`)
4. `4.7k` -> 2 rows (`R_0603_1608Metric_Pad0.98x0.95mm_HandSolder` and `R_0603_1608Metric`)
5. `4.7uF` -> 2 rows (`C_0805_2012Metric_Pad1.18x1.45mm_HandSolder` and `C_0805_2012Metric_Pad1.15x1.40mm_HandSolder`)