Part-DB/Part-DB-server #779: Unable to create/duplicate Storage Locations

CyberTeckyLLC, 2024-11-23:
Hello,

As always, I really appreciate using Part-DB as it really helps manage my electronics.  A bit of a learning curve, for sure, but worth the effort to learn and build!  

That being said, as I was creating my 'Storage Location' in Part-DB, I wanted to find an efficient way of rapidly creating/numbering storage drawer numbering.  I store my electronic components in small form-factor containers called, 'Box-All' from AideTek.  Most Box-All product can hold between 76 to 144 drawers that are designed to fit most SMD/SMT electronic components.  Here is what I use:
[image]

At the moment, I have to manually type (label) each 'Storage Location' for each Box-All container, that's 96 individual drawers that require labeling.  I thought the duplicate button at the bottom (near the create element button) would offer some form of repetitive loop back that would help rapidly number all my container + drawers.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 (for dev versions commit hash is useful]
- PHP Version: [e.g. PHP 7.4.3]
- Database Server [e.g. MySQL 5.7]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

CyberTeckyLLC, 2024-11-25:
> While it's not entirely within the software, couldn't you use something like Excel or a simple script to create the list of storage locations then use the Mass Creation tab (Storage Locations -> New Element -> Mass Creation) to do the actual creation of the locations?

I like the idea you are suggesting, thank you.  However, I'm not sure how to take the exported file from Part-DB and make it into a readable csv, yml, json, etc.  I've exported the files before, however, I just can't seem to get a grasp on organizing the data.  SQLite is a bit new for me, I suppose.

Anyone have other ideas how I should go about this would be much appreciated.  Thanks!

jbtronics, 2024-12-01:
Basically, this is what the mass creation input is for:
[image]

Basically you just have to input the elements you want to create as a (nested) list.

Maybe this could be improved by allowing to generate things like ranges automatically, but its not so easy to make a generalized solution for this.