Part-DB/Part-DB-server #782: Unsupported operand types: bool & string at IpUtils.php line 194 when accessing from link-local IP address

kolen, 2024-11-25:
**Describe the bug**
When accessing Part-DB from link-local IPv6 address, an exception in `IpUtils.php` is raised after logging in. I'm using Avahi for mDNS and its domain names resolve to link-local addresses.

**To Reproduce**
Steps to reproduce the behavior:
1. Access Part-DB through link-local IP address (i.e. `http://myhost.local` when using Avahi in some configurations)
2. Log in to Part-DB
3. A 500 Internal Server error page is displayed with the following error is logged to `(part-db dir)/var/log/prod.log`:

```json
{
    "message": "Uncaught PHP Exception TypeError: \"Unsupported operand types: bool & string\" at IpUtils.php line 194",
    "context": {
        "exception": {
            "class": "TypeError",
            "message": "Unsupported operand types: bool & string",
            "code": 0,
            "file": "/srv/http/partdb/vendor/symfony/http-foundation/IpUtils.php:194"
        }
    },
    "level": 500,
    "level_name": "CRITICAL",
    "channel": "request",
    "datetime": "2024-11-25T12:24:01.008172+00:00",
    "extra": {
        "token": {
            "authenticated": true,
            "roles": [
                "ROLE_USER"
            ],
            "user_identifier": "admin"
        },
        "url": "/partdb/public/en/login",
        "ip": "fe80::1fc4:15d8:78db:2319%enp4s0",
        "http_method": "POST",
        "server": "zubat.local",
        "referrer": "http://zubat.local:8083/"
    }
}
```

Note `ip` field in this log entry, it's http client ip address and it's a link-local address, with interface suffix, which probably causes this problem.

This exception does not occur when accessing Part-DB from `http://localhost`.

**Expected behavior**
Part-DB works with http clients from link-local IPv6 addresses

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: ce6fee1682058972c53a965db803a3cfb2f8532f
- PHP Version: PHP 8.3.14
- Database Server: sqlite

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Firefox
 - Version: 132.0.2

**Smartphone (please complete the following information):**
 - Device: N/A
 - OS: N/A
 - Browser N/A
 - Version N/A

**Additional context**

```
$ avahi-resolve-address --name zubat.local
zubat.local     fe80::b0b6:eaff:fe0d:a7f5
```

I'm using browser on the same host as server.

This file on the same http server in Part-DB directory displays `fe80::1fc4:15d8:78db:2319%enp4s0`:

```php
<?php
echo $_SERVER["REMOTE_ADDR"];
?>
```

CyberTeckyLLC, 2024-11-26:
Quote "This exception does not occur if partdb.gdpr_compliance is set to false in config/parameters.yaml."

Just to verify, are you saying that when GDPR compliance = false the issue does NOT occur?  If that is the case, I believe that would be a good thing because (in cybersecurity terms) the server should be looking for 'https' SSL/TLS certs IF GDPR compliance = true

That being said, are you trying to configure Part-DB to be accessible via local network only AND only connect via HTTPS (e.g., using SSL/TLS)?  As for my server, I'm running the same variant of Part-DB that you are using, but via a docker install on a Debian 11 VM that is running on a different host (e.g., local network connection only).  That being said, I do plan on creating self-signed SSL/TLS certs in the near future, although, I'm not sure if that would require GDPR compliance = true, or maybe it does?

Chad

jbtronics, 2024-11-26:
The only thing the GPDR mode do, is to censor the last byte(s) of IP addresses that get written into log. That way they get anonymized and they are no person relatable data anymore. That is useful if you have your Part-DB instance accessible by others. If only you access it, it doesn't matter.

The error occurs in the function that does the censoring.

I would be confident that this Is caused by the `%enp4s0` part in the IP.

CyberTeckyLLC, 2024-11-26:
> The only thing the GPDR mode do, is to censor the last byte(s) of IP addresses that get written into log. That way they get anonymized and they are no person relatable data anymore. That is useful [...]

Thanks, Jan

kolen, 2024-11-27:
> Just to verify, are you saying that when GDPR compliance = false the issue does NOT occur?

Yes.

> That being said, are you trying to configure Part-DB to be accessible via local network only AND only connect via HTTPS (e.g., using SSL/TLS)?

I'm not using TLS right now and not planning to use it, and only want Part-DB to be accessible from local network. This issue is probably unrelated to it and only caused by parsing of IP address from `$_SERVER["REMOTE_ADDR"]` which contain `%enp4s0` suffix.

CyberTeckyLLC, 2024-12-01:
> **Describe the bug** When accessing Part-DB from link-local IPv6 address, an exception in `IpUtils.php` is raised after logging in. I'm using Avahi for mDNS and its domain names resolve to [...]

Could you explain more in detail why you require Avahi for your mDNS broadcast on your local network instead of using a router that is configured with static IP and hostname assignments (e.g., MAC addresses associated with IPs)?  

That being said, I'm not sure if this would help, but I would like to share with you and the community how I have my local home network configured, and maybe it might offer alternative methods of accessing and managing your local network.

1. Router: Installed with OpenWrt firmware - It is configured to manage multiple clients, servers, and even a samba NAS on a cross-platform home network, i.e., the router handles DNS and DHCP network activity.

2. Some of my client/servers are virtual machine (VM) based and the others are physical hardware based devices.

3. Docker and VM management are centrally managed via [OpenMediaVault (OMV)](https://www.openmediavault.org/) for central web UI management and ease of use.  Part-DB was installed via docker-compose on a Debian 11 based VM which seemed to work quite well, as long as PHP version 8.3 and all required dependencies where installed.  The instructions I used were from [here](https://docs.part-db.de/installation/installation_docker.html), otherwise the missing PHP files would most likely generate a 500 Error.

Overall, the setup I used might or might not be what works for you, all I can say is that Part-DB runs great using the network and system architecture I decided to go with.  As an alternative solution, you may wish to look into Proxmox Vhttps://docs.part-db.de/installation/installation_docker.htmlE as a third solution.  

Hope this helps!

Chad

jbtronics, 2024-12-01:
This bug should be fixed with the latest commit.

closed 2024-12-01 by jbtronics

kolen, 2024-12-03:
> Could you explain more in detail why you use Avahi for your mDNS broadcast on your local network instead of using a router configured to manage static IP and hostname assignments (e.g., MAC addresses associated with IPs)?

I don't know, I thought it's most "out of the box" solution: as long as I install avahi or systemd-resolved, it just responds with hostname.local on local network without further configuration. MacOS also exports its hostname on mDNS and resolves mDNS out of the box, and iOS and Android phones resolve mDNS names too.

This mostly works okay, but I encounter issues with PHP applications for the second time, when `$_SERVER["REMOTE_ADDR"]` (or some similar cgi-style env key) returns address with interface name appended, like `%enp4s0`, and parsing code does not expect that.

Maybe I'll try setting up a normal DNS server as it's more flexible, but I have somewhat chaotic DNS resolving setup and I'm not sure if it will work with e.g. browsers having dns-over-https and similar.

>  In addition, the following image snapped from a YT video is outdated, however, may have some validity..

Seems that `.local` top-level domain is specially intended for mDNS, according to [RFC 6762](https://datatracker.ietf.org/doc/html/rfc6762).