Part-DB/Part-DB-server #784: Quick Search is Case Sensitive

peterkappelt, 2024-11-30:
Hi everyone,

I just noticed that the search field in the upper right corner is case sensitive. That seems a bit unintuitive for me, since capitalization does not convey any semantic information for part numbers most of the time IMHO.

Is this expected behaviour and/ or some kind of misconfiguration on my side? Or would y'all agree that this is unexpected and to make the search case-insentive?

**To Reproduce**
Consider a part named `B3FS-1000P`. If I type `B3` in the search field (uppercase "B"), the part is found correctly:

[image]

When entering `b3` (lowercase "b"), nothing is found:

[image]

**Expected behavior**
I'd expect the case to be case-insensitive and also show the part when searching for "b3".

**Server Side**
- Part-DB Version: 1.14.4
- PHP Version: As released in Container
- Database Server: PostgresSQL 16.6

My search options:

[image]

jbtronics, 2024-12-01:
This is something which only affected Postgres. In all other database types, all LIKE comparisons are case-insensitive by default.
Postgres however is case-sensitive, and you need to explicitly say if a comparison should be case-insensitive.
This actually not only affected the quick search, but also some other areas like Part-DB (like the normal search and filter functions).

With the latest commit this is fixed.

closed 2024-12-01 by jbtronics

peterkappelt, 2024-12-02:
Oh wow, that was quick, thank you so much!